mdhiggins/ESP8266-HTTP-IR-Blaster #298: out command for multiple LEDs does not work

g33xt4h, 2020-10-22:
I tried different gpio pins, everything is wired correctly but it always sends the command to the default LED - my command looks like this:
http://xx.xyz.net:1080/msg?code=146C7:RC6:20&out=2&pass=mypass

I looked with my phone camera which led is flashing, it's always the default one.

Thanks in advance.

mdhiggins, 2020-10-22:
Not able to reproduce this, using different outputs is something I use in my personal setup on a daily basis without issues

The code is always old and super simple and hasn't been touched in a long time and really doesn't have much room for error

```
//+=============================================================================
// Return which IRsend object to act on
//
IRsend pickIRsend (int out) {
  switch (out) {
    case 1: return irsend1;
    case 2: return irsend2;
    case 3: return irsend3;
    case 4: return irsend4;
    default: return irsend1;
  }
}
```

```
// WEMOS users may need to adjust pins for compatibility
const int pinr1 = 14;                                         // Receiving pin
const int pins1 = 4;                                          // Transmitting preset 1
const int pins2 = 5;                                          // Transmitting preset 2
const int pins3 = 12;                                         // Transmitting preset 3
const int pins4 = 13;                                         // Transmitting preset 4
const int configpin = 10;                                     // Reset Pin
```

```
IRrecv irrecv(pinr1, captureBufSize, 35);
IRsend irsend1(pins1);
IRsend irsend2(pins2);
IRsend irsend3(pins3);
IRsend irsend4(pins4);
```

That's pretty much everything. Check your setup, make sure your pin configurations aren't the problem

g33xt4h, 2020-10-22:
Hi Michael, thanks for your detailed answer - i didn't change the code, even checked the relevant parts myself.
What is the default behaviour when the pins are not wired correctly? Will it send to irsend1?

g33xt4h, 2020-10-22:
Nevermind, thanks for your help - i just realized i need to delete and redetect the device in alexa... now everything works like it should.

closed 2020-10-22 by g33xt4h