mdhiggins/ESP8266-HTTP-IR-Blaster #302: Keeping it local

SdeGeata, 2020-12-13:
Hello Michael;

   I don't have an "issue" so much as a question...

  I found this repository looking for a way to speed up the IRD transmissions to my vaporizer from Home Assistant. Presently there is a lag and often missed signals. My suspicion is that this is due to the fact that I am going through the Blynk cloud server to trigger my Wifi Blaster. Your option seems much better for me, but I am wondering if there is a way that I can send the webhooks locally, rather than going through another online service (Alexa / IFTTT) - I would rather just send over LAN and keep any speed advantage I might have.

  Can I do this? If so, how?

mdhiggins, 2020-12-13:
The device accepts any HTTP commands so if you wanted to issue those locally though something like home assistant you could probably do that (not super familiar with home assistant though). Just use the local IP address and don't open the port for external communication. You'll want to disable the Sha256 authentication though since that's tied to the Alexa skill but not a big deal if only on your local network. How you bridge those HTTP commands to voice commands is up to you. I don't know of a way that doesn't involve external networks but that's more of a question for whatever voice service you're using.

With regards to Alexa commands, there's no way to use the Alexa smart home API locally so that will need to be external

closed 2020-12-13 by mdhiggins

SdeGeata, 2020-12-13:
Thanks for the info, that will be helpful. I wasn't planning on using any voice commands either for the moment, so no bridge necessary there either. It is only local web-hooks. Thanks again!