mdhiggins/ESP8266-HTTP-IR-Blaster #303: irsent unknown signal

newbee90, 2020-12-25:
a code is only send if a name is shown, you can't send a unknown signal

serial monitor
Signal received:
One line: 8D03E0FA:UNKNOWN:21
uint16_t  rawData[41] = {922,2282, 466,1614, 480,932, 464,1632, 464,1632, 464,1632, 464,934, 464,934, 464,934, 464,932, 464,1630, 466,932, 464,1632, 466,932, 464,1632, 464,1632, 464,936, 464,1630, 488,910, 488,1608, 486};  // UNKNOWN 8D03E0FA

http://192.168.178.33/msg?code=8D03E0FA:UNKNOWN:21&pass=1234 //is not sending
but if I change UNKNOWN by sony
http://192.168.178.33/msg?code=8D03E0FA:sony:21&pass=1234 //is sending but TV is not responding
so how do I setup tehpsyc.pythonanywhere.com to send my codes

mdhiggins, 2020-12-25:
You can't send an unknown code, it means your code isn't reading correctly or isn't supported by the IR library
Only possible workaround is to send the RAW version of the code

closed 2020-12-25 by mdhiggins

newbee90, 2020-12-26:
I changed IrremoteEsp8266 to 2.7.13 am my code is not UNKNOWN anymore, not I can send code with
http://192.168.178.33/msg?code=5C2B5:METZ:19:0x3&pass=1234
but if I put im my amzn1.account.AFVDKGxxxxxxxxxxxxxxxx from tehpsyc.pythonanywhere 
I get Unauthorized, HMAC security authentication
and serial monitor says
22:45:53.132 -> Connection received - MSG
22:45:53.132 -> Failed security check, signature is too old
22:45:53.132 -> Server: 0
22:45:53.132 -> Local: 1609019152
22:45:53.132 -> MID:

mdhiggins, 2020-12-26:
Signature is too old, is your EPOCH time wrong? The security check is time sensitive so if your system clock is wrong (usually because of a DNS error causing the NTP query to fail) then the security check will fail as well

newbee90, 2020-12-27:
the difference is 2 sec
493938ms uptime; EPOCH 1609056117 / 1609056119 ( 2 )

mdhiggins, 2020-12-27:
Are you sending the command from alexa or just from your URL? Once you enable the security stuff you can't send commands over the URL anymore (unless you include all the authentication headers)

newbee90, 2020-12-27:
I send it from URL and Alexa says TV doesn't answer