mdhiggins/ESP8266-HTTP-IR-Blaster #304: is not working properly  message

newbee90, 2021-01-02:
When I say Alexa "switch TV on" that answer is TV is not working properly
[image]
[image]

mdhiggins, 2021-01-02:
You need to use the external IP address and have your port forwarded, the local address does not work as the request comes from Amazon's servers on an external network

closed 2021-01-02 by mdhiggins

newbee90, 2021-01-02:
Now it says xxxxx
[image]
[image]

 does not react

mdhiggins, 2021-01-02:
If you have a dynamic external IP address you may want to look into a dynamic DNS service

https://www.noip.com/ has a free tier that I use that works nicely

And you need to expose the local IP address of the IR blaster to port 80

To ensure you've done that correctly, you should be able to see the status page using the external IP address. If you cannot, then Alexa cannot either and you'll need to check your router settings

Some ISPs don't allow port 80 so if everything is configured correctly and its still not working try a different port. 80 is generally reserved for HTTP default and ISPs like to block that

mdhiggins, 2021-01-02:
You don't need to do anything special with your alexa IP address though (I'm assuming that's what the 192.168.178.24 entry is)

mdhiggins, 2021-01-02:
You need to specify port `80` for "port extern vergeben" for 192.168.178.33 (your IRblaster)

mdhiggins, 2021-01-02:
Alright so that looks good so far.
The port forwarding and DNS forwarding is working

The error is because your HMAC / Epoch time isn't working for some reason. This did appear to be working previously however. At the bottom of your status page where it says "EPOCH" and gives the numbers, what does it say now?

newbee90, 2021-01-02:
3162848ms uptime; EPOCH 1609620338 / 1609620340 ( 2 )
Device secured with SHA256 authentication. Only commands sent and verified with Amazon Alexa and the IR Controller Skill will be processed
Error - last authentication failed because your timestamps are out of sync, see serial output for debugging details. Timediff: 1609620265

mdhiggins, 2021-01-02:
Oh I see what's happening

You can't call the commands from the URL once you turn on the security feature, there's lots of extra data that gets transmitted from the Alexa skill to verify its a valid request

Try using the skill with your updated DNS url and see if things are working after that

Make sure you rerun "Discover devices" with Alexa after you update the URLs though

newbee90, 2021-01-02:
I delete my device and made a new search
alexa says it don't react

mdhiggins, 2021-01-02:
Screenshot what you used and send it over

mdhiggins, 2021-01-02:
And check the serial output and see if it sees the attempt to connect and perhaps why its rejecting

newbee90, 2021-01-02:
this ist the serial output
[image]
what is the other screenshot you want?

mdhiggins, 2021-01-02:
A screenshot of your device settings on the tehpsyc.pythonanywhere.com page

Also, when you set up your dynamic DNS, did you forward it to your external IP address?

newbee90, 2021-01-02:
[image]

mdhiggins, 2021-01-02:
I was able to guess your ddns address and you forwarded it to your local address WHICH WILL NEVER WORK

The whole point of ddns is that it will update as your external IP address updates. Read the documentation on that site. You can set up most routers to tell ddns when the IP changes or install software on your computer that does the same thing

https://www.noip.com/support/knowledgebase/getting-started-with-no-ip-com/

Read step 5

This is networking 101, a local IP address will not work from outside your network

newbee90, 2021-01-02:
now I have a new problem, http://glotze.xxxx.org deosn't work anymore
Alexa is ok and sending my command, as long i don't renew my IP
after renew my IP i have to start my blaster hardware new

mdhiggins, 2021-01-02:
Sounds like to just need to fix your dynamic dns and you're all set then

newbee90, 2021-01-02:
I don't know what to do anymore
[image]
[image]
[image]

mdhiggins, 2021-01-02:
Is that IP address different than the external one that your blaster device reports or are they the same?
Does going to that external IP address show the status page?
Did you try a different port? As I said before 80 is sometimes not a good choice depending on your ISP

mdhiggins, 2021-01-02:
I should be able to see your status page using that IP address and I cannot so the forwarding is not working

newbee90, 2021-01-02:
now port 222 is active and http://46.88.29.35:222/ is working
as long I don't renew my IP