mdhiggins/ESP8266-HTTP-IR-Blaster #305: external IP address and port forwarding

newbee90, 2021-01-02:
You write, "you need to use the external IP address and have your port forwarded"
my external IP adresse changes every night automatically by my router, so it won't work this way
can you give me a examble line for "Power On" in tehpsyc.pythonanywhere.com
i have port forwarded
[image]
and please do not close this issue so fast

mdhiggins, 2021-01-02:
Do not reopen the issue, I will continue support on the other issue

closed 2021-01-02 by mdhiggins

newbee90, 2021-01-02:
dns service is installed
[image]

mdhiggins, 2021-01-02:
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/304
Redirect here