mdhiggins/ESP8266-HTTP-IR-Blaster #306: WiFiManager "development" Branch Does Not Exist

PicoMitchell, 2021-02-13:
`platformio.ini` references https://github.com/tzapu/WiFiManager.git#development but when attempting to build, I get the following error because the "development" branch does not exist.
```
warning: Could not find remote branch development to clone.
fatal: Remote branch development not found in upstream origin
VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'development', 'https://github.com/tzapu/WiFiManager.git', '/Users/pico/.platformio/.cache/tmp/pkg-installing-zmryltsz']:
```

Did you mean to use the "dev_nodebug" branch? I ended up just building with the master branch and things seems to be working for me.

PicoMitchell, 2021-02-13:
Although, my ESP8266 IR Blaster does appear to be broadcasting a Wi-Fi network named like "ESP-######" which I've confirmed is coming from the ESP8266 IR Blaster board I've just set up. This is happening even after the board is successfully connected to my Wi-Fi and functioning properly. This Wi-Fi name is also not the name I set in `wifi_config_name`.

Is this related to the WiFiManger version being used? I don't remember having this problem with past ESP8266 IR Blasters I've set up.

mdhiggins, 2021-02-14:
It would appear that the development branch has been merged into master and deleted, I'll update

I think that new ESP-##### network is a side effect of WiFiManager though it may also be part of the ArduinoOTA package, not 100% sure, not something I'm creating from my blueprint though

mdhiggins, 2021-02-14:
26115e9d855c54f030f089de5f726931a4ce39d4

closed 2021-02-14 by mdhiggins