mdhiggins/ESP8266-HTTP-IR-Blaster #307: "ESP-{BOARDID}" Wi-Fi Network Broadcast Continues After Setup

PicoMitchell, 2021-02-14:
I mentioned this issue in https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/306 but since it's unrelated I've opened a new issues. Now, I think I've figured out why this was happening and how to solve/avoid it.

I'm not sure exactly why this occurred on my board, but I think this is because of WiFi mode persistence settings in WiFiManger. Seems like possibly `WIFI_AP_STA` got set on my board somehow and then persisted after `setupWifi`.

As a solution, adding `WiFi.mode(WIFI_STA);` after `// if you get here you have connected to the WiFi` seems to solve this issue for me. OR, adding `WiFi.mode(WIFI_STA);` before the `WiFiManager wifiManager;` may be a better solution so that STA mode gets persisted by WiFiManager on its own.

So, I'm not sure if this was really caused by `ESP8266-HTTP-IR-Blaster` in the first place, but thought you may want to know how to easily avoid it for any other users. A lot is going on in `WiFiManger` and `ESP8266-HTTP-IR-Blaster` and I'm not certain there would be no other consequences to adding this line. Or, there may be a better place to add this line or a better way to solve this altogether.

mdhiggins, 2021-02-14:
I'll add that line and test it out, I suspect it will probably be ok and resolve the issue

PicoMitchell, 2021-02-14:
Adding in both locations may be a better solution. I'm feeling about done with fiddling with this for now, but I just saw a case where my specified `wifi_config_name` continued to be broadcast after Wi-Fi was connected when I only had `WiFi.mode(WIFI_STA);` before the `WiFiManager wifiManager;`.

mdhiggins, 2021-02-14:
Want to drop a pull request for your changes and I'll take a look?

PicoMitchell, 2021-02-14:
Will do. I'll come back to this after a break. Think I'll do a bit more investigating first.

When logging `WiFi.getMode()` I was seeing that `WIFI_AP_STA` was still being preserved by `WiFiManager` after setting `WiFi.mode(WIFI_STA);` after `// if you get here you have connected to the WiFi`.

That made it seem like there should be a better place to put `WiFi.mode(WIFI_STA);` so that `WiFiManager` would preserve it properly on its own. But, putting it before `WiFiManager wifiManager;` didn't seem to work like I thought it would.

PicoMitchell, 2021-02-14:
Seems like putting `WiFi.mode(WIFI_STA);` before `WiFiManager wifiManager;` is the way to go. The STA mode variable does get preserved properly by `WiFiManager` that way.

`wifi_config_name` continuing to be broadcast after Wi-Fi was connected was a false positive, it just continues to stay visible to my computer for a bit (but cannot be connected to) and then goes away properly.

mdhiggins, 2021-02-15:
Merged, gonna close this for now thanks for checking this out

closed 2021-02-15 by mdhiggins