mdhiggins/ESP8266-HTTP-IR-Blaster #309: json parsing fail

SdeGeata, 2021-02-18:
Hello!

   So i finally got around to starting work on this, but am having troubles with the json stuff.

   If i use the code as is from your example, "Json as URL", it goes fine - of course i have nothing that responds to those codes. Here's the thing, once I plug my own codes in, I get parsing errors. `1FEB04F` doesn't work if I send with json, but if I send it according to the "Basic Output" format, it sends fine.  I have tried prefixing it with `0x`, that makes no difference.

   Any ideas?

SdeGeata, 2021-02-18:
A small update:

Further testing reveals to me that it is not my code throwing the error. It seems to be that it doesn't like me swapping `repeat` for `pulse`. Any idea why???

mdhiggins, 2021-02-18:
JSON parsing error means you're formatting your JSON object incorrectly

If you post what you're trying I can take a look but check your syntax

SdeGeata, 2021-02-18:
Hey! Thanks for your prompt reply!

So this is what I'm playing with right now. Its a two button sequence; the first turns the unit on, the second adjusts temperature.

`http://ArizerIRDblaster.local:80/json?pass=PASSWORD&plain=[{"type":"nec","data”:”1FE7887”,”length":32,”address”:0x80},{“type":"nec","data”:”1FEB04F”,”length":32,”address”:0x80,”pulse”:10,”pdelay”:100}]`

Still getting errors.

mdhiggins, 2021-02-18:
Couple things

- You don't have to use quotes around the key for ArduinoJson if you want to make your URLs shorter
- You forgot quotes around the address which are required (simple integers don't need this but anything else does)
- Whatever character you're using for quotes in some places isn't a normal quote, I would use a proper programming text editor cause `"` is different than `”` which will parse differently

```
http://ArizerIRDblaster.local:80/json?plain=[{type:"nec",data:"1FE7887",length:32,address:"0x80"},{type:"nec",data:"1FEB04F",length:32,address:"0x80",pulse:10,pdelay:100}]&pass=PASSWORD
```

closed 2021-02-18 by mdhiggins

SdeGeata, 2021-02-18:
> You don't have to use quotes around the key for ArduinoJson if you want to make your URLs shorter

Ah, well that is good to know. I took the example off of the page but it seemed quotation mark heavy.

> You forgot quotes around the address which are required (simple integers don't need this but anything else does)

Also good to know.

> Whatever character you're using for quotes in some places isn't a normal quote, I would use a proper programming text editor cause " is different than ” which will parse differently

Yeah, i discovered that my computer was automatically changing the straight brackets and replacing them with curly ones. A programming app doesn't seem to make a difference - I use Nova. I did find the option to turn it off in my settings though so all is well.

Thank you for your help and good work.