mdhiggins/ESP8266-HTTP-IR-Blaster #310: Error external IP, no NTP response & amazon user id

ICookiee, 2021-02-18:
Hello, 

I have several questions:

At first, my program doesn't work like it should. I tried to configure the AP, and now it connects successfully to the WLAN. But there are several errors:

`ESP8266 IR Controller
15:11:10.621 -> Config pin GPIO10 set to: 1
15:11:10.669 -> mounted file system
15:11:10.669 -> reading config file
15:11:10.669 -> opened config file
15:11:10.669 -> {"hostname":"IRRemote","passcode":"Pass","port_str":"80","user_id":"xx","ip":"10.0.1.10","gw":"10.0.1.1","sn":"255.255.255.0","dns":"10.0.1.1"}
15:11:10.715 -> parsed json
15:11:10.715 -> *WM: [3] allocating params bytes: 20
15:11:10.715 -> *WM: [2] Added Parameter: hostname
15:11:10.715 -> *WM: [2] Added Parameter: passcode
15:11:10.715 -> *WM: [2] Added Parameter: port_str
15:11:10.715 -> *WM: [2] Added Parameter: user_id
15:11:10.715 -> Setting static WiFi data from config
15:11:10.715 -> *WM: [1] AutoConnect 
15:11:10.715 -> *WM: [2] Connecting as wifi client... 
15:11:10.715 -> *WM: [3] STA static IP: 10.0.1.10
15:11:10.715 -> *WM: [2] Custom static IP/GW/Subnet/DNS 
15:11:10.715 -> *WM: [2] Custom static DNS 
15:11:10.715 -> *WM: [1] STA IP set: 10.0.1.10
15:11:10.715 -> *WM: [1] Connecting to SAVED AP: xx
15:11:10.715 -> *WM: [3] Using Password: xx
15:11:10.761 -> *WM: [3] WiFi station enable 
15:11:10.761 -> *WM: [3] enableSTA PERSISTENT ON 
15:11:10.761 -> *WM: [1] connectTimeout not set, ESP waitForConnectResult... 
15:11:10.902 -> *WM: [2] Connection result: WL_CONNECTED
15:11:10.902 -> *WM: [3] lastconxresult: WL_CONNECTED
15:11:10.902 -> *WM: [1] AutoConnect: SUCCESS 
15:11:10.902 -> *WM: [1] STA IP Address: 10.0.1.10
15:11:10.902 -> WiFi connected! User chose hostname 'IRRemote' passcode 'Pass' and port '80'
15:11:10.948 -> *WM: [3] freeing allocated params! 
15:11:10.948 -> *WM: [3] unloading 
15:11:10.948 -> WiFi configuration complete
15:11:10.948 -> Local IP: 10.0.1.10
15:11:10.948 -> DNS IP: 10.0.1.1
15:11:10.948 -> URL to send commands: http://IRRemote.local:80
15:11:10.948 -> ArduinoOTA started
15:11:10.948 -> MDNS http service added. Hostname is set to IRRemote.local:80
15:11:10.948 -> HTTP Server started on port 80
15:11:10.948 -> Starting UDP
15:11:10.948 -> Local port: 8888
15:11:10.948 -> Waiting for sync
15:11:10.948 -> Transmit NTP Request
15:11:12.915 -> Turning off the LED to save power.
15:11:20.931 -> time.google.com: 255.255.255.255
15:11:22.471 -> No NTP Response :-(
15:11:32.705 -> Error retrieving external IP
15:11:32.705 -> HTTP Code: -1
15:11:32.705 -> connection failed
15:11:32.705 -> External IP address request took 10219 ms
15:11:32.751 -> Warning, user_id appears to be in the wrong format, security check will most likely fail. Should start with amzn1.account.***
15:11:32.751 -> Epoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings
15:11:32.751 -> 18022
15:11:32.751 -> Invalid EPOCH time, security checks may fail if unable to sync with NTP server
15:11:32.751 -> Ready to send and receive IR signals
`

What am i doing wrong? Where can i find my amazon user id?
Thank you!

mdhiggins, 2021-02-18:
What is your router's IP address?

Amazon ID from the readme:
"This unique ID can be found at the bottom of the tehpsyc.pythonanywhere.com page"

ICookiee, 2021-02-18:
IP: 192.168.178.1

Oh, sorry for that. Guess I missed this one in the readme.

mdhiggins, 2021-02-18:
https://medium.com/@sadatnazrul/basics-of-ip-addresses-in-computer-networking-f1a4661ea85c

Read up on the basics of networking, particularly the subnets section

The short answer why things aren't working is that you defined all your IP address settings wrong and setup the device outside of your own network

The gateway and dns server should likely be the same IP address as your router
The static IP address you choose should be on the same subnet as your router
Your subnet mask of 255.255.255.0 is probably ok

You'll need to reset your board and enter this information correctly

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster#force-wifi-reconfiguration
That's how you reset those settings and boot the board back into configuration mode (reflashing the blueprint won't be enough)

ICookiee, 2021-02-18:
I totally missed the setup part in the .ino also...
Works now. Thank you!

closed 2021-02-18 by ICookiee