mdhiggins/ESP8266-HTTP-IR-Blaster #313: router port closing

chrisncc1701, 2021-03-08:
Hi, my router port closes after a few hours, i don't know if it's inactivity, if i log on to the blaster localy it sometimes reopens, i disabled wifi_set_sleep_type(LIGHT_SLEEP_T) in setup and increased delay to 400 in the loop, but don't really know what i'm doing, i'm using :- ARUINO JSON V 6.17.3,  WIFIMANAGER V 2.0.3 alpha,  NTPCLIENT V 3.2.0, IRremote V 2.7.1, the 8266 core is at v 2.7.4, i tried dropping to V 2.4.0 but it won't compile, having read previous issues the latest wifimanager is incorporated in the IDE, please help

chrisncc1701, 2021-03-08:
forgot to mention my ip settings are :-1: 
char static_ip[16] = "192.168.1.37";
char static_gw[16] = "192.168.1.1";
char static_sn[16] = "255.255.255.0";
char static_dns[16] = "192.168.1.1";

mdhiggins, 2021-03-08:
You're not supposed to modify those parameters those are meant to store the data from WiFiManager (you shouldn't need to edit the blueprint at all for network configuration)

You no longer need to drop to 2.4.0, that was an issue with older version of WiFiManager

Why won't it compile? What's the error?

Consider using PlatformIO if you're having trouble setting up your build environment since that will do it for you and handle version control

chrisncc1701, 2021-03-10:
Hi, sorry, i thought you had to change those parameters to match my router, i have put them back to 10.0.1.10 etc, it is now working fine for 2 day's, the compiling issue is not a problem as i don't have to drop to 2.4.0, thank you for your help, the unit does crash if i try to remote log in on its external ip address and port but it's not really an issue, just wish i was good enough to know how it all worked?, i'll comment in the next few days

mdhiggins, 2021-03-10:
Alright gonna close this. External requests shouldn't crash anything but keep an eye on the serial output when you make the requests that results in a crash and see if it gives any insight

closed 2021-03-10 by mdhiggins