mdhiggins/ESP8266-HTTP-IR-Blaster #314: Exception (3)

philipptrenz, 2021-03-14:
Hey there, 

tried to use this project on a D1 Mini using PlatformIO. Compiles and uploads fine, but after that I get the following error within the serial monitor in an continuous loop:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

ESP8266 IR Controller
Config pin GPIO5 set to: 1
mounted file system

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40100c91 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4002ce98 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff9d0 end: 3fffffc0 offset: 0190
3ffffb60:  00000001 00000000 00000002 00000040  
3ffffb70:  500ffc10 00010001 ffffffff 3ffe91e8  
3ffffb80:  3fff2090 3fff1e30 00000020 40100f37  
3ffffb90:  3ffe91e7 3fff20c8 00000000 4022b0ac  
3ffffba0:  4022c6e8 3fff20c8 3ffffeb4 4023b299  
3ffffbb0:  4022c6e8 3fff20c8 3ffe95e1 4023b284  
3ffffbc0:  3fff2090 00000013 3fff20c8 402224ae  
3ffffbd0:  3fff2090 3fff20c8 3ffffde0 40222504  
3ffffbe0:  402318d0 3fff20c8 3ffffde0 4020b540  
3ffffbf0:  00000000 00000000 3fff3d44 3fff3d49  
3ffffc00:  3fff3d4c 3fff3cdc 3fff3cec 3fff3cec  
3ffffc10:  3fff3cf4 3fff3d08 3fff3d08 3fff3d0c  
3ffffc20:  3fff3d20 3fff3d20 402331e8 00000000  
3ffffc30:  402331e8 00000000 402331e8 00000000  
3ffffc40:  402331e8 00000000 402331e8 00000000  
3ffffc50:  402331e8 00000000 402331e8 00000000  
3ffffc60:  feef5035 6e2d6f6e fe007465 86efeffe  
3ffffc70:  feefef00 feefeffe 80efeffe feefef00  
3ffffc80:  feefeffe 80efeffe feefef00 feefeffe  
3ffffc90:  80efeffe feefef00 feefeffe 80efeffe  
3ffffca0:  feefef00 feefeffe 80efeffe 0002bf20  
3ffffcb0:  00000000 00000000 00000000 00000000  
3ffffcc0:  00000001 00505345 feefeffe 83efeffe  
3ffffcd0:  feefef00 00000000 00000000 00000000  
3ffffce0:  000007d0 00010000 00000000 0050ef00  
3ffffcf0:  feef0701 ffffffff 00000000 00000000  
3ffffd00:  01000001 00010101 01010001 fe010001  
3ffffd10:  3ffe8c25 3ffe8c25 feefef00 feefeffe  
3ffffd20:  80efeffe 3fff3d24 000b000f 00efeffe  
3ffffd30:  feef0000 00007530 feefef00 feefef00  
3ffffd40:  feefeffe 80efeffe feef0000 00000000  
3ffffd50:  feef0100 00000000 00000000 00000005  
3ffffd60:  00000000 feefef01 3a6d772a feefef00  
3ffffd70:  84efeffe 00000002 feefef02 3fff20c8  
3ffffd80:  40207490 3fff1d8c 402318f8 402318e4  
3ffffd90:  feefeffe feefeffe 00000000 feefeffe  
3ffffda0:  402053a8 3fff20c8 402318d0 402318bc  
3ffffdb0:  feefeffe feefeffe 00000000 feefeffe  
3ffffdc0:  feefeffe feefeffe 00000000 feefeffe  
3ffffdd0:  feefeffe feefeffe 00000000 feefeffe  
3ffffde0:  402053a8 feefeffe 402318d0 402318bc  
3ffffdf0:  feefeffe feefeffe feefeffe feefeffe  
3ffffe00:  feefeffe feefeffe feefeffe feefeffe  
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
3ffffe30:  feefeffe feefeffe feefeffe feefeffe  
3ffffe40:  feefeffe feefeffe feefeffe feefeffe  
3ffffe50:  feefeffe feefeffe feefeffe feefeffe  
3ffffe60:  3ffffee0 00000000 312306dd 402306dd  
3ffffe70:  3ffffee0 feefeffe feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe 3fff20c8  
3ffffe90:  4022c6e8 3fff20c8 3ffffedf 4022c6f4  
3ffffea0:  4022c6e8 3fff20c8 3ffffedf 4022cb0d  
3ffffeb0:  00000031 3ffe9229 3ffe9276 4022ce9d  
3ffffec0:  00000014 00000001 3ffe8c25 feefeffe  
3ffffed0:  3ffe95e3 00000000 000a0d78 402306dd  
3ffffee0:  3ffe95e3 00000000 5700203a 402306dd  
3ffffef0:  00000009 40230c3d ffffffff 3fff2214  
3fffff00:  4022c6e8 3fff20c8 3ffe95e1 4022c6f4  
3fffff10:  4022c6e8 3fff20c8 3ffe95e1 4022cb0d  
3fffff20:  00000000 00000001 3fff20c8 4022cd8c  
3fffff30:  3fffdad0 00000000 3fff20c8 3fff2214  
3fffff40:  3fffdad0 00000000 3fff20c8 4020bbd3  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff2214  
3fffffa0:  3fffdad0 00000000 3fff21d4 4022f3a4  
3fffffb0:  feefeffe feefeffe 3ffe856c 40100419  
<<<stack<<<
```

What can I do to trace the issue down?

philipptrenz, 2021-03-14:
Traced it down using https://github.com/me-no-dev/EspExceptionDecoder
Decoded Exception:

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x40100c91
EXCVADDR: 0x4002ce98

Decoding stack results
0x4023b299: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x4023b284: operator delete[](void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_opv.cc line 3
```

philipptrenz, 2021-03-14:
Could also test using a NodeMCUv2 Lolin board, same behaviour.

philipptrenz, 2021-03-14:
Switched over to ArduinoIDE and now it works, not sure what went wrong using PlatformIO

closed 2021-03-14 by philipptrenz

mdhiggins, 2021-03-14:
Could you share what build options you used in the Arduino IDE that did work? I'm guessing its probably something in there that wasn't playing nice

Mainly looking for board, flash size, lwip variant, and vtables

I might just need to specify I different environment for d1_mini_lite and lolin boards in platform io

mdhiggins, 2021-03-14:
b385f22cda0bfb2a6cfe1c0ac32acf8d5ce726fd

I added some additional board environments

philipptrenz, 2021-03-16:
[image]

These are the Arduino settings I used, in PlatformIO I used the following configuration:

```
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
lib_deps =
  IRremoteESP8266@^2.7
  https://github.com/tzapu/WiFiManager.git
  ArduinoJson@^6.17
  Time@^1.6
  https://github.com/jjssoftware/Cryptosuite.git
monitor_speed = 115200
```

mdhiggins, 2021-03-16:
Does the new d1_mini environment that's included now work for you?