mdhiggins/ESP8266-HTTP-IR-Blaster #315: Router port closing

chrisncc1701, 2021-03-16:
Hi, I still have the problem of the router port closing after a few hours/days, (issue #313) i have changed the port number and all libraries are up to date, the 8266 becomes unavailable due to the port being closed but if i log on locally the serial output say's "connection received" the port reopens and all is ok

mdhiggins, 2021-03-16:
If the local requests are working and the serial output is fine then this is a networking problem you'll need to investigate. 

I've never heard of a router closing a port on its own, that would be bizarre. Is the port forwarding setting still in your router configuration?

Is it possible your external IP address isn't just changing? Lots of ISPs use a dynamic external IP address that may change at varying frequencies

chrisncc1701, 2021-03-16:
the external IP address isn't changing and the forwarding settings on the router are still ok and there's nothing on the router logs, it just appears that the router thinks there's no connection so closes the port, but i know what your saying about being able to log on locally, as far as the esp is concerned it's still connected, the forwarding rules are TCP/UDP, don't suppose there is any way of restarting the ESP when the port closes, i don't understand how i can log on locally when the port is set to 303, i appreciate it could not be a problem with this project but any help from anywhere would be appreciated

mdhiggins, 2021-03-16:
This is definitely odd behavior and something that would need to be addressed at the router level, though I can't say I've ever heard of a router doing that. Having the ESP restart isn't really a great solution as the device itself isn't doing a deep sleep or anything, it should remain connected and listening continuously while it's on. Nothing on the serial output suggests the device has crashed or is rebooting or anything does it? Does the uptime counter reset or anything on the local page?

What kind of router?

chrisncc1701, 2021-03-16:
It is a Huawie HG633 router, i am going to leave a laptop on monitoring serial for when it happens again and also check the uptime, at the moment i don't think the service provider will take any notice

mdhiggins, 2021-03-16:
Try some of the solutions here

https://community.talktalk.co.uk/t5/Product-Archive/HG633-port-forwarding-CCTV/td-p/2189059/page/2

Looks like this is a specific problem with your router

mdhiggins, 2021-03-16:
You can also try adding this line at 508 and see if it helps `client.keepAlive();`

```
  Serial.println("WiFi connected! User chose hostname '" + String(host_name) + String("' passcode '") + String(passcode) + "' and port '" + String(port_str) + "'");

  client.keepAlive();

  // Save the custom parameters to FS
```

chrisncc1701, 2021-03-17:
Hi, I tried that command but the device went into a continuous sort of restart, and i have looked at various forums about my router, and it looks like the port closes with inactivity but i don't know why the time interval varies sometimes it runs for over a day, i don't think talktalk are going to be much help, iv'e included the serial output withe the command on line 508, i have noticed that the parameters set in AP didn't look as though they where saved with client.keepalive(); on line 508

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: hostname
*WM: [2] Added Parameter: passcode
*WM: [2] Added Parameter: port_str
*WM: [2] Added Parameter: user_id
Setting static WiFi data from config
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP: 10.0.1.10
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom static DNS 
*WM: [1] STA IP set: 10.0.1.10
*WM: [1] Connecting to SAVED AP: TALKTALKC596CD
*WM: [3] Using Password: A8W6THXD
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 10.0.1.10
WiFi connected! User chose hostname '' passcode '' and port '80'

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40231f97 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffa90 end: 3fffffc0 offset: 0190
3ffffc20:  3fff2174 3ffffed4 3ffffe3c 4020b80b  
3ffffc30:  00000000 00000000 3fff441c 3fff441f  
3ffffc40:  3fff4420 3fff43f4 3fff4404 3fff4404  
3ffffc50:  3fff440c 3fff4418 3fff4418 40232f98  
3ffffc60:  00000000 40232f98 00000000 40232f98  
3ffffc70:  00000000 40232f98 0a01000a 40232f98  
3ffffc80:  0101000a 40232f98 00ffffff 40232f98  
3ffffc90:  0101000a feef5035 6e2d6f6e fe007465  
3ffffca0:  86efeffe feefef00 feefeffe 80efeffe  
3ffffcb0:  feefef00 feefeffe 80efeffe feefef00  
3ffffcc0:  feefeffe 80efeffe 0002bf20 00000000  
3ffffcd0:  00000000 00000000 00000000 00000001  
3ffffce0:  00505345 feefeffe 83efeffe feefef03  
3ffffcf0:  00000000 00000000 00000000 000007d0  
3ffffd00:  00010000 00000000 00500000 40100407  
3ffffd10:  ffffffff 00000001 00000001 01000001  
3ffffd20:  00010101 01010001 3fff0100 3ffe8c35  
3ffffd30:  3ffe8c35 40103f00 3fff02e0 80efeffe  
3ffffd40:  40100001 00007530 feefef00 feefef00  
3ffffd50:  feefeffe 80efeffe feef0100 00000000  
3ffffd60:  40100100 00000000 00000004 00000005  
3ffffd70:  3fff451c feef0301 3fff2470 40207450  
3ffffd80:  00000000 40231668 40231654 feefeffe  
3ffffd90:  feefeffe 00000000 feefeffe 40205378  
3ffffda0:  40231640 40231640 4023162c 3fffc278  
3ffffdb0:  401029f8 00000000 00000022 feefeffe  
3ffffdc0:  4022d3d7 00000000 00000010 ffffffff  
3ffffdd0:  4022d46b 00000000 0000000a 40003000  
3ffffde0:  0101000a 80231690 4023167c 00000006  
3ffffdf0:  00000006 00000000 00000000 3ffffedf  
3ffffe00:  0000000a 0000000a 3fff2470 00000030  
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe feefeffe  
3ffffe30:  feefeffe feefeffe 3fff4700 3fff302c  
3ffffe40:  80efeffe feefeffe feefeffe feefeffe  
3ffffe50:  feefeffe feefeffe feefeffe 3ffe91b1  
3ffffe60:  3ffe92c5 3fff4564 0000003c 00000001  
3ffffe70:  3ffe8c35 feefeffe feefeffe feefeffe  
3ffffe80:  3ffe9239 3ffe92b7 3fff4554 00000006  
3ffffe90:  00000001 3ffe8c35 3ffe92ae 3ffe92a5  
3ffffea0:  3fff4534 00000014 00000001 3ffe8c35  
3ffffeb0:  3ffeb458 3ffe927d 3fff44fc 00000014  
3ffffec0:  00000001 3ffe8c35 3fff4600 000c000f  
3ffffed0:  80fe95ea 3fff4800 0040004f 80230a45  
3ffffee0:  3ffe9500 00000000 80570020 40230a45  
3ffffef0:  00000009 40230fa5 ffffffff 40232f98  
3fffff00:  0101000a 40232f98 00ffffff 40232f98  
3fffff10:  0101000a 40232f98 0a01000a 4022d07d  
3fffff20:  00000000 00000001 3fff2470 4022d2fc  
3fffff30:  3fffdad0 00000000 3fff2470 3fff25bc  
3fffff40:  3fffdad0 00000000 3fff2470 4020bb8f  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff25bc  
3fffffa0:  3fffdad0 00000000 3fff257c 4022f728  
3fffffb0:  feefeffe feefeffe 3ffe8570 401011c9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

ESP8266 IR Controller
Config pin GPIO10 set to: 1
mounted file system
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: hostname
*WM: [2] Added Parameter: passcode
*WM: [2] Added Parameter: port_str
*WM: [2] Added Parameter: user_id
Setting static WiFi data from config
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP: 10.0.1.10
*WM: [2] Custom static IP/GW/Subnet/DNS 
*WM: [2] Custom static DNS 
*WM: [1] STA IP set: 10.0.1.10
*WM: [1] Connecting to SAVED AP: TALKTALKC596CD
*WM: [3] Using Password: A8W6THXD
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 10.0.1.10
WiFi connected! User chose hostname '' passcode '' and port '80'

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

mdhiggins, 2021-03-17:
Try putting it at line 950 instead

```
  irrecv.enableIRIn();

  client.keepAlive();

  Serial.println("Ready to send and receive IR signals");
```

mdhiggins, 2021-03-17:
Compiles and runs just fine on my test device using line 950

chrisncc1701, 2021-03-18:
thank you, have done that, will comment in a few day's, meantime i'll try get talktalk to send another router haha

mdhiggins, 2021-03-18:
Feel free to continue discussion

closed 2021-03-18 by mdhiggins

chrisncc1701, 2021-04-02:
Hi, Just to let you know, talktalk have been very helpful and sent me a new router a DG8041w and the help desk been on the case, the port now stays open, so don't know if the HG633 was faulty or just the way it did port forwarding, so big thanks for your help and great project

mdhiggins, 2021-04-02:
Glad you got it working!