mdhiggins/ESP8266-HTTP-IR-Blaster #316: Unauthorized access

DomoYop, 2021-03-18:
Hi !

First of all, Thanks for your incredible job with on this project !
really amazing !

but i get an error with this latest version [https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/commit/f14efc74ba89a2fdaa1e5bb0c3958bdb3c1ea346](url)
i cant access to the Web UI ( http://ir_controller.local/ or http://192.168.194.131/ ), i always get an "Unauthorized access"

here is my log (no obvious error):

> 
ESP8266 IR Controller
Config pin GPIO4 set to: 1
mounted file system
reading config file
opened config file
{"hostname":"IR_Controller","passcode":"password","port_str":"80","user_id":"amzn1.account.xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx","ip":"192.168.194.131","gw":"192.168.194.1","sn":"255.255.255.0","dns":"192.168.194.1"}
parsed json
*wm:[2] Added Parameter: hostname
*wm:[2] Added Parameter: passcode
*wm:[2] Added Parameter: port_str
*wm:[2] Added Parameter: user_id
Setting static WiFi data from config
*wm:[1] AutoConnect
*wm:[2] Connecting as wifi client... 
*wm:[2] Custom static IP/GW/Subnet/DNS
*wm:[2] Custom static DNS
*wm:[1] STA IP set: 192.168.194.131
*wm:[1] Connecting to SAVED AP: Wifi_Local 
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS
*wm:[1] STA IP Address: 192.168.194.131
WiFi connected! User chose hostname 'IR_Controller' passcode 'password' and port '80'
WiFi configuration complete
Local IP: 192.168.194.131
DNS IP: 192.168.194.1
URL to send commands: http://IR_Controller.local:80
ArduinoOTA started
MDNS http service added. Hostname is set to IR_Controller.local:80
HTTP Server started on port 80
Starting UDP
Local port: 8888
Waiting for sync
Transmit NTP Request
time.google.com: 216.239.35.0
Receive NTP Response
External IP: xx.xx.xx.xx
External IP address request took 77 ms
No errors detected with security configuration
EPOCH time obtained for security checks
Ready to send and receive IR signals
Turning off the LED to save power.
Connection received endpoint '/'
Unauthorized access
Connection received endpoint '/'
Unauthorized access

i have try it on 2 different board, with Arduino IDE and Platform.io, same result
i'm surely missing something...

thanks for help !

mdhiggins, 2021-03-18:
Security has been tightened in the recent versions so that the user portal requires the passcode now

That being said this update I just pushed

6e5d38edc85aae0ca2338f20fa38c77ba8093e10

will allow local network traffic to ignore the passcode similar to the HMAC security

DomoYop, 2021-03-22:
Works as expected !

thank you for your help :D

closed 2021-03-22 by DomoYop