mdhiggins/ESP8266-HTTP-IR-Blaster #318: getting error no ntp response

jayrawatrj, 2021-04-24:
WiFi configuration complete
  Local IP: 10.0.1.10
  DNS IP: 10.0.1.1
  URL to send commands: http://blaster.local:80
  ArduinoOTA started
  MDNS http service added. Hostname is set to blaster.local:80
  HTTP Server started on port 80
  Starting UDP
  Local port: 8888
  Waiting for sync
  Transmit NTP Request
  Turning off the LED to save power.
  time.google.com: 255.255.255.255
  No NTP Response :-(
  Error retrieving external IP
  HTTP Code: -1
  connection failed
  External IP address request took 10276 ms
  Ready to send and receive IR signals

jayrawatrj, 2021-04-24:
i am not getting the ip address for "time.google.com" as it is showing 255.255.255.255 every time. but when i am running example code from Time library named as "TimeNtp_ESP8266" that code is working fine.

mdhiggins, 2021-04-24:
Local IP: 10.0.1.10
DNS IP: 10.0.1.1

Are these values valid and on your network subnet? Can't access the internet if they're not

jayrawatrj, 2021-04-24:
these values are declare in the code itself.

jayrawatrj, 2021-04-24:
may be i got the solution

closed 2021-04-24 by jayrawatrj

mdhiggins, 2021-04-24:
They are the default values and should be overwritten by the options specified in WiFi manager

jayrawatrj, 2021-05-14:
ok. got it. Thanks for response