mdhiggins/ESP8266-HTTP-IR-Blaster #319: Only works once per device after power up

The-Arsonist, 2021-05-14:
I've had my controller working for several years. Recently, it will only respond once per device after a power up. For example, I can tell my TV volume to turn up, and it works as expected, but If I ask the TV volume to turn up again, it won't do anything, and the ESP8266 IR Controller web page shows no signal sent. However, if I ask it to turn the receiver volume up, it will do it, even though they are sending the same signal (but separate devices in the controller setup) but after that one time, the receiver will now not send IR signals either.

mdhiggins, 2021-05-14:
Show me the command you are trying to send, include the full URL (though you can censor your IP address and pass/security information)

The-Arsonist, 2021-05-14:
http://externalip:port/msg?code=5EA158A7:NEC:32&device=Receiver&pass=hunter2&out=2

mdhiggins, 2021-05-14:
Try leaving out the &device=Receiver part

Specifying the device is only for saving states for toggle commands you its probably causing it to ignore subsequent commands as a result

The-Arsonist, 2021-05-14:
Ah, thanks. That's probably my issue, I forgot the device is already being specified in the setup page. Must have been left over when I copied my code from the power state command

closed 2021-05-14 by mdhiggins

TheMCMan, 2021-05-14:
That's an awesome collaboration :-)