mdhiggins/ESP8266-HTTP-IR-Blaster #323: WiFiManager crashes on save when setting up AP information.

stresser1, 2021-06-29:
I've been trying to make this work all week. When I use the aurdiono IDE I can't find the libraries, so I had to guess until it compiled. but once it compiles it doesn't load, just reboots. So I cloned the repository and used PlatformIO. it loads and I can log into the webserver, but when I save the AP details, it crashes. NOTE I had to use the decoder in the Aurdino IDE and it wouldn't let me until I compiled the code in it... which didn't run on the device...  you can see attached esp 8266 IRBlaster oddessey.txt for more detail on what I tried and the error messages in serial monitor.

Don't rule out I'm an idiot and missed something critical. I'm an application and backend programmer for PCs, this is my first home builder level microcontroller... the closest I've come to this stuff was allen bradley PLCs back in the 90s, and that was just rudimentary programs, nothing near as complicated as this, and you only needed the vendor ide, not this layered stack of stuff.

Decoded exception

`Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40239ad8
EXCVADDR: 0x06680000

Decoding stack results
0x40230080: String::changeBuffer(unsigned int) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\WString.cpp line 191
0x402302df: String::String(String const&) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\WString.cpp line 46
0x40230311: String::operator=(char const*) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\WString.cpp line 261
0x40208dc8: esp8266webserver::ESP8266WebServerTemplate ::_parseRequest(WiFiClient&) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/Print.h line 46
0x4025882d: umm_info(void*, bool) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\umm_malloc/umm_info.c line 122
0x40100554: esp_schedule() at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\core_esp8266_main.cpp line 128
0x40211547: loop() at C:\Users\lukerediger\Desktop\IR blaster\IRController/IRController.ino line 1721
0x402338df: std::_Function_handler   >::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 136
0x40100fba: free(void*) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\umm_malloc\umm_malloc.cpp line 595
0x40228c7d: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSAnswer_TXT(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1642
0x40224fbb: esp8266::MDNSImplementation::MDNSResponder::_announce(bool, bool) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 1589
0x40100554: esp_schedule() at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\core_esp8266_main.cpp line 128
0x40228fb0: esp8266::MDNSImplementation::MDNSResponder::_prepareMDNSMessage(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&, IPAddress) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 260
0x40222f68: WiFiManager::handleWifiSave() at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/WString.h line 277
0x4022918a: esp8266::MDNSImplementation::MDNSResponder::_sendMDNSMessage_Multicast(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 141
0x4022d594: fs::File::File(std::shared_ptr , fs::FS*) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/shared_ptr_base.h line 619
0x4022d5a0: fs::File::peek() at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/shared_ptr_base.h line 1328
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x4022eaf8: MD5Builder::toString() const at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\MD5Builder.cpp line 94
0x4021096d: setupWifi(bool) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 330
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x40207ad4: configModeCallback(WiFiManager*) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0/new line 175
0x40233b84: std::function ::function(std::function  const&) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 498
0x40233b70: std::function ::function(std::function  const&) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 498
0x40205448: Uri::~Uri() at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\libraries\ESP8266WebServer\src/Uri.h line 16
0x40233b5c: std::function ::function(std::function  const&) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 587
0x40233b48: std::__shared_count(__gnu_cxx::_Lock_policy)0>::~__shared_count() at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/shared_ptr_base.h line 317
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x40233b5c: std::function ::function(std::function  const&) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 587
0x40233b48: std::__shared_count(__gnu_cxx::_Lock_policy)0>::~__shared_count() at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/shared_ptr_base.h line 317
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x4022d594: fs::File::File(std::shared_ptr , fs::FS*) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/shared_ptr_base.h line 619
0x402358dc: read_uleb128 at /workdir/repo/gcc-gnu/libgcc/unwind-pe.h line 141
0x4022e880: std::_Function_handler   >::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) at c:\users\lukerediger\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.3-gcc10.3-9bcba0b\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h line 303
[... cut ...]
0x40210f27: setup() at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266/WString.h line 350
0x402314d8: loop_task(ETSEvent*) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\core_esp8266_main.cpp line 215
0x401004b1: cont_check(cont_t*) at C:\Users\lukerediger\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.1\cores\esp8266\cont_util.cpp line 46`

[esp8266 ir blaster odessy.txt](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/6730630/esp8266.ir.blaster.odessy.txt)

stresser1, 2021-06-29:
also, if I ever get this working, I can do up a new wiki for you, it's rather out of date compared to what is here, and the .bas file is no longer available.

mdhiggins, 2021-06-29:
Looks like they just released ESP8266 Arduino Core 3 which breaks a bunch of stuff, just use version 2 for now, I just updated the platform.io files to explicitly indicate version 2

mdhiggins, 2021-06-29:
Wiki was just an old relic from version 1 before this was even an Arduino project, I deleted that page the readme covers everything

Also the passcode thing is is not your wifi password but an optional passcode you can set to authorize commands via the URL

stresser1, 2021-06-29:
Thanks for the quick updates! I recloned and the compile is failing now. 

xtensa-lx106-elf-g++: error: src\IRController.ino.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\IRac.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\IRrecv.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\IRsend.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\IRtext.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\IRtimer.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\IRutils.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Airwell.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Aiwa.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Amcor.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Argo.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Carrier.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Coolix.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Corona.cpp.o
Compiling .pio\build\nodemcuv2\libe38\IRremoteESP8266\ir_Daikin.cpp.o
*** [.pio\build\nodemcuv2\src\IRController.ino.cpp.o] Error 1

mdhiggins, 2021-06-29:
I would clean your build directory and try again

[image]

stresser1, 2021-06-29:
I realized the wiki was old when I was going for the .BAS file as the wiki is where I started, lol, then when I couldn't find the .bas I started looking around and realized you'd made massive updates, and started over.

ok, so I feel really noobish right now, I'm not seeing where I would find that icon in platformIO and I'm not having luck with google

mdhiggins, 2021-06-29:
[image]

stresser1, 2021-06-29:
omg, thank you, my lower bar is blue...

it's uploaded, and working now, thank you so much!

closed 2021-06-29 by stresser1

mdhiggins, 2021-06-29:
Looks like I might have prematurely assumed v3 was the problem, I cleaned by build folder and now v3 compiles too so I just reverted that last commit that forced v2, feel free to update

stresser1, 2021-06-29:
.... I'll give it a try on another one tomorrow as my wife is making "bed time now!" eyes at me. I will let you know how it goes

stresser1, 2021-06-29:
Good Morning!

I cleared the build folder, I recloned the project to a new folder, uploaded blink to the one that kept crashing, imported the newly cloned version again, and compile/uploaded it. and it worked.

No idea why it wasn't working before... perhaps I had a bad reference to the core somewhere? Did I just need to clear the build folders, even though I'd never used it before?

Thanks for your assistance. I will make up a word doc with detailed instructions from start to finish with pictures etc so you can send it to any other fish out of water like myself in the future.

BTW, do you know which libraries I should be importing into Arduino IDE as per my txt file, some of them seem to be name changed to cover both 8266, 32 and AT? That way I can include that info as well.

Cheers!

mdhiggins, 2021-06-29:
If you're using platform IO you shouldn't even need the ArduinoIDE installed, platform IO is an independent program that will handle dependencies

stresser1, 2021-06-29:
understood, I haven't figured out how to run serial monitor on Platform yet, nor how to tell it to wipe the entire memory when uploading, but I had in Arduino IDE, so I kept using it to condition the ModeNCUs. 

I want the info to update instructions since you seem to have a set for each. 

I'm going to make it for myself as an aide de memoire, and because I have some guys at work who are also interested in this project for their own home automation, that are even less tech savvy than me, and they'll need help.

stresser1, 2021-06-29:
I also didn't know if I could install the exception decoder in platformio

mdhiggins, 2021-06-29:
[image]

Checkmark is build
Right arrow is upload to device
Trash is clean build folder
Plug is serial monitor

Open PlatformIO IDE Terminal and type `pio run --target erase` to fully wipe device, though in reality you shouldn't need to do this often

You might need to specify an `upload_port` in your platformio.ini file if its not autodetected

Exception decoder is the one exception, you'll need the IDE for that as far as I know

mdhiggins, 2021-06-29:
```
lib_deps =
  IRremoteESP8266@^2.7
  https://github.com/tzapu/WiFiManager.git
  ArduinoJson@^6.17
  Time@^1.6
  https://github.com/jjssoftware/Cryptosuite.git
```

Those are all the dependencies as referenced by the platformio ini file. The 3 not on github are all in the Library Manager and don't need special installation if you wanted to to use the IDE. The other 2 you'll need to download from Github and manually add to the library

stresser1, 2021-06-29:
fantastic!

I'll put together the doc and get you the PDF to critique.

mdhiggins, 2021-06-29:
I wouldn't go too crazy, the mqtt branch is going to overhaul lots of stuff soon and add a whole new layer of functionality

Feel free to try it out
https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/tree/mqtt-dev

stresser1, 2021-06-29:
Thanks! once I get my guide put together, I will definitely check that out... I've got a co worker champing at the bit looking to see what I've been working on. 

I've discovered a lot of odd issues with my boards as they are not what I thought, and appear to be odd ball V3s, which are 1.0 /12Es not 0.9/12s... I've recompiled for 1.0 and re uploaded and they seem to be working the same.

I am using the Platformio Monitor, but the command to erase isn't erasing the flash, while doing it in aurdino ide does when I set it to clear all flash and upload blink. I just had an issue where I uploaded 1.0 compiled into my second blaster and it did the reboot loop thing I had before, and did it again after doing the erase in PIO and re uploading. So I did the aurdino blink upload, then uploaded the compiled 1.0 and it worked fine. There must be more options, and I'll look into that later.

My LEDs are also not spec'd so I'm adjusting the current limiting resistance down to approach my hand remote. I'm down to 2.7ohms, and it's close enough. but I suspect I'm being current limited from my USB port on my computer... I might promptly burn out the ir LED once I plug it into a wall wart.

All of these trials and tribulations I'm logging/noting in my doc. I'm mostly an tech writer for end user SOPs these days, so it's not a burden and helps me think about what I'm doing anyway. writing it in more detail than I did the txt I attached earlier is keeping me relaxed and slowing me down to actually think about what I'm doing, and hopefully will limit the hand holding you've been doing for me, which is very appreciated.

that said, is the ongoing conversation in here ok, or would you rather I stop?

mdhiggins, 2021-06-29:
Feel free to keep discussion going

The current and resistance you pull off the board should be the same regardless, you're pulling it off the USB controller of the board which is voltage regulated

stresser1, 2021-06-29:
Thanks, I'm new to Git and I based on how fast you were answering questions I was worried I might be being a pest.

I am using the transistor set up, it's pretty much required to get these LEDs to light up enough to be useable. I am hoping to have these some distance from the TVs and receivers so I wanted to take the best approach as possible.

My boards are not standard, so I don't trust that the usbv pin is regulated. The Vin pin you use is only outputting 0.25v, but my boards put a USB V and USB G on the 2 reserved pins, so I relocated the LED in and transistor emitter to those pins. I suspect there is a voltage regulator after the vin pin blocking using it as a vout because the breakout board I bought with them has a 6-24v power jack, which is why I don't trust this will last if I plug it into a 1a switching charger. If they don't I'll swap them out with ones I can get specs for, they came with my receivers and looked the part so I'm just using them for now based on how they do.

Also shorting SD3 isn't triggering wifi config reset, though it is outputting 3v, so honestly at this point I think I'm lucky these things work at all.

BTW, clever use of the if elseif else statements to control authentication. It took me a few minutes to realize local IPs won't be HMAC checked if I set bypassLocalAuth to false. I probably would have written a much larger if statement including cases that actually aren't needed.

I'm not intending to use HMAC because it needs to reach external to talk to amazon if I understand correctly, and since I'm going to go through a local hosted webpage to trigger the functions rather than Amazon/apple etc. (I might look into Smart things later though)

mdhiggins, 2021-06-29:
No trouble

For the HMAC thing if you don't intend to use Amazon just leave the Amazon userid blank. That will keep the hmac system disabled and then just don't forward the ports to your external network and it should be fairly secure. 

The security functions are actually much cleaner and more consolidated on the MQTT branch.

MQTT is nice because it allows external services like Amazon without the need for opening a port so should be even more secure than the HMAC option. Still testing that out though I've got all my personal devices running on it without issue for about a month now.

stresser1, 2021-06-29:
nice! I assume a purely self contained network will still be possible?

mdhiggins, 2021-06-29:
Yeah the existing HTTP based framework is still there. Just adding MQTT as another option.

You can also use your own MQTT broker locally or just use local only HTTP commands

stresser1, 2021-06-30:
first draft of the noob guide for beginners like myself
[IRBlasterNoobGuide_Draft_1.0.pdf](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/6737146/IRBlasterNoobGuide_Draft_1.0.pdf)

mdhiggins, 2021-06-30:
Very detailed, you've been working hard on this

Couple notes though

> bypassLocalAuth to false to make it require a pass code to work (note this is all in the clear so this is not a particularly good security method, however, there will be a new version of this out soon, that does much better security

The security methods aren't really changing, HMAC Sha256 is still going to be the standard via HTTP. bypassLocalAuth just means you trust those on your local network to access the device and want to grant permission. This is helpful since once you enable HMAC you really can't send any commands easily without also generating all the authentication headers so building and testing via the URL before setting up an Alexa skill is challenging. I leave bypassLocalAuth on for my personal network since its a trusted network but anything that comes from an external source will require the HMAC headers.

MQTT will offer an alternate approach but that's not to say the security methods used for the HTTP method or flawed and I wouldn't want to imply that, MQTT is just a different solution with its own advantages/disadvantages and is separate from the HTTP protocol

> Once the AP has been set, if you want to go back to the configuration menu, short pin GPIO10(SD3) to ground. (Note this doesn't work on my v3 boards, I have to reload blink in Arduino IDE then re upload from platform IO.)

This is probably because the V3 board keeps a live current going through the default configpin (10 in this case) unlike the V2 version (https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/issues/142)
If you can find an alternative available pin that does allow ground I can add an alternate configuration to the blueprint for the V3 boards similarly to how I have configurations for the WeMOS variants. **Just tell me what pin ends up working.**  This will also alleviate your need to wipe the filesystem like you keep needing to do. You just need to find a pin who's default state is HIGH but can be set to LOW when connected to ground. I don't have a V3 board to test on.

> This software cannot handle rapid repeated requests... it will crash without an exception. The web server just stops responding, and you'll still see updates for the NTP server for example in the serial monitor, but the web server doesn't come back up until you reset. I suspect the automatic reboot might reset it but I don't bet on it. So, to send multiple commands, use the repeat code word

The webserver should be able to handle repeated requests without crashing and I can hammer mine without issue, so this might be a problem on your end or with your board. The repeat command does however give better more reliable timing since its all packaged together and not dependent on the latency of the webserver / connection

mdhiggins, 2021-06-30:
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

References for the V2 for comparison regarding the pinouts and states at boot

stresser1, 2021-06-30:
thanks, I will update it with this information later, and as I build my web page I'll add notes to that, then start incorporating other stuff like if I tap into smart things, which I've been using for my samsung TVs. since I don't have any of the other platforms, my ability to write a guide is limited.

I will play around with the boards to find a better config pin, that may be the issue with the web server choking on rapid fire requests.

mdhiggins, 2021-06-30:
Sounds good, keep me posted

stresser1, 2021-06-30:
GPIO 0,2,3,4,5,9,10,12,13,14 set to 1 at start up

I'm not sure why that's important, is this built into wifimanager, and it stops monitoring the reset pin if it's pulled up? You'd think it would just wait a while after boot, to check to see if the pin is grounded.

1 caused the device to boot in a bad state... pressing reset caused garbled info to go into the serial monitor in a reset AP state, connecting to the AP caused a few odd error messages

4, caused some weird reboot thing when set transmit to 1.

6,7,8,11 - reset loop  
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v0007d820
~ld

ESP8266 IR Controller

15,16 pulled down at start up triggering wifi config every boot.

14 is reciever
4 is transmit 1
5 is transmit 2
12 is transmit 3
13 is transmit 4

So it looks like I'm out of luck when it comes to a config pin

stresser1, 2021-06-30:
since the tick sub uses GPIO1 the weird state makes sense now

stresser1, 2021-06-30:
I'm an idiot... I have to short the pin to ground, then hit reset in order for the AP setup to start... I will re test this later... I'm going to be away for a while, so it will probably be monday. Have a good night.

stresser1, 2021-07-05:
const uint16_t  pinr1 = 14;                                          // Receiving pin
const uint16_t  pins1 = 4;                                           // Transmitting preset 1
const uint16_t  configpin = 12;                                       // Reset Pin
const uint16_t  pins2 = 5;                                           // Transmitting preset 2
const uint16_t  pins3 = 16;                                          // Transmitting preset 3
const uint16_t  pins4 = 13;    

As requested, this works to reset the V3 into AP mode for reconfigure

stresser1, 2021-07-05:
The web server jamming up seems to be gone now too. I don't know if it's moving the config pin, or using Arduino IDE and Blink to wipe the board prior to uploading the new firmware.

I had to wipe it again after changing the pins as it was crash/rebooting when trying to access the AP.

stresser1, 2021-07-05:
[IRBlasterNoobGuide_Draft_1.1.pdf](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/files/6761150/IRBlasterNoobGuide_Draft_1.1.pdf)

stresser1, 2021-07-05:
here is a picture of my web page I'm using, the idea is it's a favorite on my phone browser, I'm going to add "macro" buttons that are javascripts that call sequences. I have the calls from the buttons going to an iframe below the remotes.

[image]

stresser1, 2021-07-05:
I came up with another use for the IR receiver, if you added a table of look up codes, you could program buttons on one remote to rebroadcast as if another sent them. 

I.E. one of the reasons I'm doing this is to have a fully custom, no frills universal remote for my rec room. 

The TV is piping sound into the surround sound receiver, but the two don't talk, even though they both support the same protocols (brand mis match) so if I could put a look up table into the main loop, that allowed me to specify when it receives the code for volume up on Samsung, it immediately sends out the JVC receiver volume up code, same with mute/vol down and alt power. 

I think I prefer the path I'm taking, but thought I would share.