mdhiggins/ESP8266-HTTP-IR-Blaster #324: Reversed IR Blast sent to Home Assistant

Ian-Zz, 2021-07-28:
This isn't an issue. Is there any way possible I could have this AWESOME  ESP8266-HTTP-IR-Blaster code send a received IR blast from a remote control to my instance of Home Assistant? I would love to be able to set up a remote to do Hassio buttons and switches etc.

Btw... I do have another Node-Mcu running ESP-Home doing just this thing... But it's in the garage. Would like it for living room now too. Just wondering. Thanks

mdhiggins, 2021-07-28:
You could just add whatever code you need to talk to Home Assistant in the receiver functionality of the blueprint, but I don't personally use home assistant or have any idea what would need to be sent or for what purpose but the handling of received codes is in the main loop

```
void loop() {
  ArduinoOTA.handle();
  server->handleClient();
  decode_results  results;                                        // Somewhere to store the results

  if (irrecv.decode(&results) && !holdReceive) {                  // Grab an IR code
    Serial.println("Signal received:");
    fullCode(&results);                                           // Print the singleline value
    dumpCode(&results);                                           // Output the results as source code
    copyCode(last_recv_4, last_recv_5);                           // Pass
    copyCode(last_recv_3, last_recv_4);                           // Pass
    copyCode(last_recv_2, last_recv_3);                           // Pass
    copyCode(last_recv, last_recv_2);                             // Pass
    cvrtCode(last_recv, &results);                                // Store the results
    last_recv.timestamp = now();                                  // Set the new update time
    last_recv.valid = true;
    Serial.println("");                                           // Blank line between entries
    irrecv.resume();                                              // Prepare for the next value
    digitalWrite(ledpin, LOW);                                    // Turn on the LED for 0.5 seconds
    ticker.attach(0.5, disableLed);
  }
  delay(200);
}
```

Just add whatever logic you need inside the if statement

closed 2021-07-28 by mdhiggins

Ian-Zz, 2021-07-29:
Thanks a bunch for the quick response mdhiggins.
I just learned today of the mqtt edition/pull of this code.
Trying it that way for now. Just need to get the topics tidy some.
Thanks again. Very Much.

mdhiggins, 2021-07-30:
So you know aside from the old pull request there's also a much more up to date MQTT branch

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/tree/mqtt-dev