mdhiggins/ESP8266-HTTP-IR-Blaster #325: mgtt on local network with mosquitto

hakha4, 2021-08-16:
Just found out that there is a mqtt version for connecting to Alexa.  Tried to configure for local network but complaining about userid (for Alexa). Is there a way to bypass Alexa (unavaliable in Sweden) and connect to mosquitto on an rpi  ? My aim is to send ir from Home assistant via mqtt integration.
BTW thank's for a awesome program!
Any hint's appreciated
Regards Håkan

mdhiggins, 2021-08-16:
I actually just started using and setting up home assistant this week, I'll update the branch to support this once I've got more figured out. HA seems solid but lots to learn as a noob. 

Since I'm new to home assistant, how exactly were you planning to configure things so I can mirror your setup and try and get things working ASAP

hakha4, 2021-08-16:
I'm quite new to home assistant also but beginning to manage the basics now. In HA there is a integration called MQTT where one can subscribe to a topic and my plan is to try implement that. Since I'm not planning to  expose data outside my local network an option not to have TSL security on Mosquitto  would be first choice (seems to be messy to set up, at least for a non proffesional).  Mosquitto resides in docker on a raspberry pi.

I have a regular ESP8266 IR-blaster that works like a charm and a 'brutal' way to send IR from HA is to create a button and in click event choose 'URL' and put in same info as in a web browser. Works but the problem is that IR-blaster webpage opens and you loose focus  from HA, so one option could be  if one could tell IR-blaster to send IR without open the webpage but MQTT maybe a more universal approach.
There must be a number of users that use both HA and IR-blaster so someone might have input on this.
Regards Håkan

Ian-Zz, 2021-08-17:
I am also very interested in an mqtt friendly version of this blaster. I tried the mqtt version and I just never could get it loaded and working well if at all.

I mentioned it in a previous "issue" but would love to use IR received commands to activate switches and scripts etc in Home Assistant. It seems mqtt would make this nearly a no brainer. Aka my style. 

Nice too would be able to have pretty good control over the pub and sub topics & messages. Looking forward to mqtt properly with this nifty as heck blaster code.

mdhiggins, 2021-08-24:
Quick aside
For the mqttdev branch, the user_id is just a prefix that gets applied to the MQTT listening subscription, so if you're not using amazon you can just put any prefix string in that field to satisfy the requirement

So for my own home-assistant setup I've been trying a number of options to get seamless integration using the media-player entity and I've found that local HTTP requests have worked better for me than MQTT. MQTT on the ESP8266 seems a bit unreliable and I've had lots of memory issues even with attempts at aggressive memory management, lots of random crashes

I'm using HTTP commands but running the mqttdev branch since this contains some general improvements all around including device states that can be strings instead of just ints

I'll share my config below to show how I've implemented things using universal media player

configuration.yaml excerpt
```
# REST sensor that reads if the TV is on using the device state memory of the IRController
rest:
  - scan_interval: 40
    resource: http://10.0.1.102:8083/state
    sensor:
      - name: Rest Gym TV
        value_template: "{{ value_json.states.state }}"

# Command template to send RESTFul commands to the IRController
rest_command:
  gymtv:
    url: http://10.0.1.102:8083/json?simple=1
    method: POST
    payload: "{{ payload }}"
    content_type: "application/json; charset=utf-8"

# Universal media player wrapper
media_player:
  - platform: universal
    name: Gym TV
    children:
      - media_player.gym_roku
    commands:
      turn_on:
        service: script.gym_tv_on_routine
      turn_off:
        service: script.gym_tv_off_routine
      volume_up:
        service: rest_command.gymtv
        data:
          payload: "[{data:'490',type:'SONY',length:12,out:3}]"
      volume_down:
        service: rest_command.gymtv
        data:
          payload: "[{data:'C90',type:'SONY',length:12,out:3}]"
      volume_mute:
        service: rest_command.gymtv
        data:
          payload: "[{data:'290',type:'SONY',length:12,out:3}]"
    attributes:
      state: sensor.rest_gym_tv
```

and here are the gym_tv_on_routine and gym_tv_off_routine

```
gym_tv_on_routine:
  sequence:
  - service: rest_command.gymtv
    data:
      payload: '[{type:''SONY'',data:''A90'',length:12,out:3,device:''state'',state:''on''}]'
  - service: media_player.select_source
    target:
      entity_id: media_player.gym_tv
    data:
      source: Plex - Free Movies & TV
  - service: homeassistant.update_entity
    target:
      entity_id: sensor.rest_gym_tv
  mode: single
  alias: Gym TV On Routine
gym_tv_off_routine:
  alias: Gym TV Off Routine
  sequence:
  - service: rest_command.gymtv
    data:
      payload: '[{type:''SONY'',data:''A90'',length:12,out:3,device:''state'',state:''off''}]'
  - service: media_player.media_stop
    target:
      entity_id: media_player.guest_bedroom
  - service: homeassistant.update_entity
    target:
      entity_id: sensor.rest_gym_tv
  mode: single
```

I'm setting the device `state` in the on and off commands as part of the payload and reading that back every 40 seconds as a sensor
I'm trigger a sensor refresh using `homeassistant.update_entity` any time an on or off command is sent to make sure this is up to date
The on routine also opens Plex since this is the only app I use on the roku, but this would be optional
For simple commands like volume no script is needed and instead you can see I'm calling the rest_command service directly

The device `media_player.gym_roku` is a roku that's plugged into the gym_tv which when the TV is on will report current playback information

[image]

mdhiggins, 2021-08-28:
2b7f7992d5eb203136467c82f38888010fa05423

New update includes first draft at publishing the device state to the mqtt server
Not supported by the Alexa skill since that's read only so intended only for private mqtt brokers

You'll need to set `const bool mqtt_publish = true;` at the top in the user settings variables (defaults to false)

Should automatically publish that device states to the mqtt broker when they are changed by a command or if the USB power monitoring feature detects a change

Let me know how that works

hakha4, 2021-08-31:
Good news!  I've been busy with work last couple of days so I hadn't have a chance to test yet, but I'll get back ASAP.

closed 2022-01-25 by mdhiggins