mdhiggins/ESP8266-HTTP-IR-Blaster #326: Pioneer codes aren't being sent.

ESurge, 2021-09-12:
Hi there, I've recently switched my A/V receiver from a Sony to a Pioneer.

I've been using this project to operate the Sony receiver successfully for some time now. However, upon switching to the Pioneer A/V Receiver (VSX-1028), I can't seem to get operations to work properly.

I don't have the original remotes for either Sony or Pioneer devices, and therefore I'm using a Logitech Harmony remote to send the commands to the receiver and ESP8266.

For diagnostics, I'm using an IR camera/filter on my phone camera and when the IR LED is triggered, it's quite apparent.

Below are the commands I receive from the Harmony remote on the ESP8266. When I attempt to send any of the received commands, formatted in the same way, **the IR LED does not trigger.** Also, using the raw data doesn't always work either.

Power On:
/msg?code=A55A58A7A55A58A7:PIONEER:64
[image]

Power Off:
/msg?code=A55AD827A55AD827:PIONEER:64
[image]

Volume Up:
/msg?code=A55A50AFA55A50AF:PIONEER:64
[image]

Volume Down:
/msg?code=A55AD02FA55AD02F:PIONEER:64
[image]

Any ideas? Thanks.

Edit: After writing all this, I was playing around with the remote again, and verifying that what I wrote was accurate.

I noticed that the ESP reported one of the signals as a NEC:32 signal with the first half of the Pioneer code (A55A58A7:NEC:32 and address 0xa5).

Using this formatting (/msg?code=A55A58A7:NEC:32&address=0xa5) for Power On/Off and Vol Up/Down, the code worked. However, some other functions still don't work with this method (Menu/Exit buttons, as well as various Input buttons).

Menu/Exit buttons provide the following codes:

Menu:
/msg?code=A55AD926A55A6B94:PIONEER:64 with address 0xa59b

Exit:
/msg?code=A55ABA45A55A906F:PIONEER:64 with address 0xa55d

What I've noticed is that with the Power On/Off and Vol Up/Down codes, the first 8 digits repeat for the second 8 digits. This is not the case with the Menu/Exit buttons. This repeating code is probably why the NEC:32 code ended up working.

Any help as to why the PIONEER:64 codes aren't being sent would be great.

Thanks again.

mdhiggins, 2021-09-12:
Try using the new mqttdev branch, it contains lots of updates to the base code as well and should support pioneer better than the current live version. The MQTT stuff is all optional and not relevant if you're not using it

ESurge, 2021-09-12:
Hey thanks for the quick reply. I've updated to the mqtt-dev branch and everything is working great now.

Thanks again!

closed 2021-09-12 by ESurge

mdhiggins, 2021-09-12:
Good stuff. That should be merged into the master branch soon