mdhiggins/ESP8266-HTTP-IR-Blaster #327: Unable to sent requests by alexa

bm-a, 2022-01-01:
Hello i am able to successfully open the web portal and also receive and send IR requests locally but not over alexa i have used ip 192.168.1.42 and the port as 80 and also configured port forwarding please take a look at it below.

Also i am not able to see the esp in dhcp client list in the router settings.

Could you assist me and help me to find out the issue.

[image]

bm-a, 2022-01-01:
21:05:13.325 -> ESP8266 IR Controller
21:05:13.325 -> Config pin GPIO10 set to: 1
21:05:13.325 -> mounted file system
21:05:13.325 -> reading config file
21:05:13.325 -> opened config file
21:05:13.325 -> {"hostname":"IR BLASTER","passcode":"","port_str":"667","user_id":"","ip":"192.168.1.12","gw":"192.168.1.1","sn":"255.255.255.0","dns":"117.96.122.81"}
21:05:13.325 -> parsed json
21:05:13.325 -> *wm:[2] Added Parameter: hostname
21:05:13.325 -> *wm:[2] Added Parameter: passcode
21:05:13.325 -> *wm:[2] Added Parameter: port_str
21:05:13.372 -> *wm:[2] Added Parameter: user_id
21:05:13.372 -> Setting static WiFi data from config
21:05:13.372 -> *wm:[1] AutoConnect 
21:05:13.372 -> *wm:[1] No Credentials are Saved, skipping connect 
21:05:13.372 -> *wm:[2] Starting Config Portal 
21:05:13.372 -> *wm:[2] Disabling STA 
21:05:13.372 -> *wm:[2] Enabling AP 
21:05:13.372 -> *wm:[1] StartAP with SSID:  IR Controller Configuration
21:05:13.888 -> *wm:[2] AP has anonymous access! 
21:05:14.358 -> *wm:[1] AP IP address: 192.168.4.1
21:05:14.358 -> Entered config mode
21:05:14.358 -> 192.168.4.1
21:05:14.358 -> IR Controller Configuration
21:05:14.358 -> *wm:[1] Starting Web Portal 
21:05:14.358 -> *wm:[2] HTTP server started 
21:05:14.358 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
21:05:14.405 -> *wm:[2] Portal Timeout In 180 seconds
21:05:25.278 -> *wm:[2] <- Request redirected to captive portal 
21:05:26.029 -> *wm:[2] <- HTTP Root 
21:05:26.076 -> *wm:[2] <- Request redirected to captive portal 
21:05:26.123 -> *wm:[2] <- Request redirected to captive portal 
21:05:26.969 -> *wm:[2] <- HTTP Wifi 
21:05:26.969 -> *wm:[2] WiFi Scan SYNC started 
21:05:29.132 -> *wm:[2] WiFi Scan completed in 2188 ms
21:05:29.132 -> *wm:[1] 2 networks found
21:05:29.132 -> *wm:[2] AP:  -65 HOME WIFI
21:05:29.132 -> *wm:[2] 
21:05:29.508 -> *wm:[2] <- Request redirected to captive portal 
21:05:29.650 -> *wm:[2] <- Request redirected to captive portal 
21:05:31.434 -> *wm:[2] <- Request redirected to captive portal 
21:05:43.263 -> *wm:[2] Portal Timeout In 163 seconds
21:06:13.239 -> *wm:[2] Portal Timeout In 133 seconds
21:06:43.262 -> *wm:[2] Portal Timeout In 103 seconds
21:07:13.247 -> *wm:[2] Portal Timeout In 73 seconds
21:07:43.248 -> *wm:[2] Portal Timeout In 43 seconds
21:07:55.184 -> *wm:[2] <- HTTP WiFi save  
21:07:55.184 -> *wm:[2] Parameters 
21:07:55.184 -> *wm:[2] -------------------- 
21:07:55.184 -> *wm:[2] hostname: IR BLASTER
21:07:55.231 -> *wm:[2] passcode: pass
21:07:55.231 -> *wm:[2] port_str: 667
21:07:55.231 -> *wm:[2] user_id: amzn1.account.xxxxxxxxxxxxxxxxxxxxxxxxxx
21:07:55.231 -> *wm:[2] -------------------- 
21:07:55.231 -> *wm:[2] processing save 
21:07:57.207 -> *wm:[2] Connecting as wifi client... 
21:07:57.254 -> *wm:[2] Custom static IP/GW/Subnet/DNS 
21:07:57.254 -> *wm:[2] Custom static DNS 
21:07:57.254 -> *wm:[1] STA IP set: 192.168.1.42
21:07:57.254 -> *wm:[1] Connecting to NEW AP: HOME WIFI
21:07:57.348 -> *wm:[1] connectTimeout not set, ESP waitForConnectResult... 
21:08:02.519 -> *wm:[2] Connection result: WL_CONNECTED
21:08:02.519 -> *wm:[1] Connect to new AP [SUCCESS] 
21:08:02.519 -> *wm:[1] Got IP Address: 
21:08:02.519 -> *wm:[1] 192.168.1.42 
21:08:02.519 -> Should save config
21:08:02.519 -> *wm:[2] shutdownConfigPortal 
21:08:02.519 -> *wm:[2] restoring usermode STA
21:08:03.553 -> *wm:[2] wifi status: WL_CONNECTED
21:08:03.553 -> *wm:[2] wifi mode: STA
21:08:03.553 -> *wm:[2] configportal closed 
21:08:03.553 -> *wm:[1] config portal exiting 
21:08:03.553 -> WiFi connected! User chose hostname 'IR BLASTER' passcode 'pass' and port '667'
21:08:03.553 ->  config...
21:08:03.553 -> {"hostname":"IR BLASTER","passcode":"pass","port_str":"667","user_id":"amzn1.account.xxxxxxxxxxxxxxxxxxxxxxx","ip":"192.168.1.42","gw":"192.168.1.1","sn":"255.255.255.0","dns":"192.168.1.1"}
21:08:03.553 -> Writing config file
21:08:03.646 -> Config written successfully
21:08:03.646 -> WiFi configuration complete
21:08:03.646 -> Local IP: 192.168.1.42
21:08:03.646 -> DNS IP: 192.168.1.1
21:08:03.646 -> URL to send commands: http://IR BLASTER.local:667
21:08:03.646 -> ArduinoOTA started
21:08:03.646 -> MDNS http service added. Hostname is set to IR BLASTER.local:667
21:08:03.646 -> HTTP Server started on port 667
21:08:03.646 -> Starting UDP
21:08:03.646 -> Local port: 8888
21:08:03.646 -> Waiting for sync
21:08:03.646 -> Transmit NTP Request
21:08:04.399 -> time.google.com: 216.239.35.8
21:08:04.587 -> Receive NTP Response
21:08:05.622 -> External IP: 1xx.1xx.xx.xx
21:08:05.622 -> External IP address request took 1047 ms
21:08:05.622 -> No errors detected with security configuration
21:08:05.622 -> EPOCH time obtained for security checks
21:08:05.622 -> Ready to send and receive IR signals
21:08:05.622 -> Turning off the LED to save power.

MY DEFAULT gateway is 192.168.1.1
and Subnet mask is 255:255.255.0

mdhiggins, 2022-01-01:
Are you telling Alexa to use your external IP address? You can't use 192.168.xxx.xxx for Alexa as that's your local IP. You need to use your external IP

bm-a, 2022-01-01:
yes i am using http://xxx.1xx.xx.xx:80/json?plain=[{data:[7734,10940,14,9000,10854,16,8932,11038,14,8942,11034,12,9032,10996,14,16,9108,10880,16,9108,10834,14,9124,10856,12,9118,10870,14,12,9078,10932,9082,10916,9122],type:'raw',khz:38}] external ip like this above.(this is just a raw random data)

bm-a, 2022-01-01:
can i try and send web requests (for testing purpose) without password if i delete password in wifi manager settings and also remove my amazon id from there.

mdhiggins, 2022-01-01:
Yes that should work fine. That URL should return an access denied message too if you try and access it over a browser without credentials

mdhiggins, 2022-01-01:
I would also check and make sure your IP address isn't behind a double NAT which can cause lots of trouble. 

Check and see what your routers external IP versus what's reported on a website like whatismyip.com

bm-a, 2022-01-01:
How can i find if i am behind nat or how can i check my routers ip

> I would also check and make sure your IP address isn't behind a double NAT which can cause lots of trouble.
> 
> Check and see what your routers external IP versus what's reported on a website like whatismyip.com

How can i find if i am behind nat or how can i check my routers ip

bm-a, 2022-01-01:
Yes i think i am on nat as my routers ip is 192.168.137.1 in ipconfig on cmd where as my external ip is different

mdhiggins, 2022-01-02:
Yeah that would suggest you're behind a double NAT which means you cannot forward ports from your external IP address unfortunately, your options here are to see if that can be changed by your ISP (some ISPs will give you a direct external IP upon request, although sometimes that costs extra). Other option would be to look into setting up your own MQTT server and using the MQTT-dev branch of the script which does not require port forwarding but needs an MQTT server as an intermediate

bm-a, 2022-01-02:
I checked with my ISP it costs extra per month can I use an esp8266 as an intermediate  server?

mdhiggins, 2022-01-02:
You'll need some kind of server outside of your local network as an intermediate which is where the MQTT server comes into play, there are probably some free options out there that you could try though I'm less experienced in this area so you'll need to look into it more

closed 2022-01-04 by bm-a