mdhiggins/ESP8266-HTTP-IR-Blaster #328: VERY Insecure

bwosborne2, 2022-01-18:
This needs to be secured, at a minimum, with https SSL encryption before exposing directly to the Internet

It might be better to route this through a secure, Internet hardened reversy proxy such as an nginx server

rocket357, 2022-01-18:
https://www.bhbreptiles.com/collections/milk-snakes/products/pueblan-milksnake-2021?variant=170347092

bwosborne2, 2022-01-18:
That should be listed earlier, in the primary instructions. Many users looking for a solution will not drill down that deep. 

I just corrected somebody on a forum looking to expose 50 televisions to the Internet using this .

rocket357, 2022-01-18:
An alternative would be to document using it with Google Assistant (via Home Assistant) which does not require exposing the IRBlasters to the internet.

TheMCMan, 2022-01-18:
Mine is secured by Raspberry Pi proxy.  But this is a project add-on.  Mine
also works with Google assistant.

bwosborne2, 2022-01-18:
> Google Assistant (via Home Assistant)

You have HA parsing the http string? The user I corrected was on the HA forum.

Apparently the developer wants to support Alexa though.

rocket357, 2022-01-18:
> You have HA parsing the http string? The user I corrected was on the HA forum.

My use case, arguably, is a simple one.  I have a few IRBlasters attached to lights (just on/off), so the sequences are pretty simple.

The TL;DR is to configure a Home Assistant to have a switch with an automation attached to toggle it back and forth, then call that from a fullfillment URL in the Google Assistant Add-On.  There will need to be something like this for the automation to call:

`rest_command:
  master_led_off:
    url: "http://10.120.128.1:8080/msg?code=F7C03F:NEC:32&address=0xef00"
    method: put
  master_led_on:
    url: "http://10.120.128.1:8080/msg?code=F740BF:NEC:32&address=0xef00"
    method: put`

mdhiggins, 2022-01-18:
Security limitations are discussed at length in the readme and there are security options in place within the limits of the hardware. The MQTT beach also offers an additional solution without opening the port for those that wish to provide their own MQTT server. Reverse proxies or using an intermediate such as HA are also options.

closed 2022-01-18 by mdhiggins