mdhiggins/ESP8266-HTTP-IR-Blaster #329: Range of IR blaster

daknightuk, 2022-01-19:
So I followed this diagram exactly
100ohm resistor
2n2222 transistor
1k resistor
IR333-A

However the range of my IR blaster is about 3 metres. I need about 5m
I'm using the 5v Vin pin 30 
am wondering if there is actually 5v

I've been able to use an IR LED between GND and GPI04 and it works with no transistor, resistor but still doesn't give the required range.

Any ideas on how I can get the range up?

David

mdhiggins, 2022-01-19:
5V should be the highest output, can check with a voltmeter if you're getting that, some knock off or v3 boards might not give you full 5V, not much you can do about that though other than get a different board

Otherwise, you have essentially 2 options to get more brightness
1. Get a different LED
2. Use a lower resistance

If you combine a bright LED that can handle more current, you should get more brightness

Connecting directly to GPIO4 will give you pretty low current and brightness, the whole reason for using the transistor and connecting to 5V is to push as much as possible
The 1k resistor won't affect brightness, that's just for the transistor to control the circuit

I use a 10-ohm resistor and these LEDs https://www.amazon.com/gp/product/B00ULB0U44/ which can handle higher bursts of current for short stints on some of my devices that need longer range

closed 2022-01-19 by mdhiggins

daknightuk, 2022-01-20:
Hi Michael,
I'm using a V2 board, https://www.az-delivery.uk/products/nodemcu
I couldn't get the same IR LED as you from anywhere so I looked on this website where the datasheet was and found out its actually an Everlight IR333-A https://cdn-shop.adafruit.com/datasheets/IR333_A_datasheet.pdf
After I messaged I found a circuit diagram which had a similar setup but no R1 1k ohms on the base of the transistor.
I tried it just to see what it would do and strangely I got brighter output and more range, not as much as I would like but I'm guessing that this will be down to the voltage as you say. I will have to get a voltmeter on Vin and see what its reading.
I'll try a 10ohm and see whether it makes it brighter

David

mdhiggins, 2022-01-20:
I wouldn't think the transistor resistor would make a difference (unless you maybe have it connected to the wrong leg of the transistor) but I'm no electrical engineer

mdhiggins, 2022-01-21:
I'm wrong

https://forum.arduino.cc/t/how-to-calculate-resistor-to-a-base-of-a-2n2222/206798/22

Decreasing both resistors is probably beneficial here

mdhiggins, 2022-01-21:
Looks like you should just max the current coming off the GPIO in addition to maybe lowering your LED resistor

12mA appears to be the max for the GPIO pins so adjust your resistor accordingly

daknightuk, 2022-01-24:
@mdhiggins So it appears that there are slight variations on the ESP8266, I have 2  circuits (one excluding all resistors) which works great has a long range and one which I had to add the resistors back in as it Popped the IR LED. 
Must be a sign of poor manufacturing or quality control, I've ordered the LED you suggested and will use it with a 10 ohm resistor.