mdhiggins/ESP8266-HTTP-IR-Blaster #330: 'USECPERTICK' was not declared in this scope

vilioren56, 2022-02-14:
Hello everyone, tell me how to solve the problem, using the latest version of the library 2.8.1
`'USECPERTICK' was not declared in this scope`
maybe it should be replaced with another variable?

mdhiggins, 2022-02-14:
Fixed in #59 in like 2017, are you using a very old version of the blueprint?

vilioren56, 2022-02-21:
Arduino 1.6.9 and your latest libraries 2.8.1
Please help, a very acute problem

mdhiggins, 2022-02-21:
Post the full error cause I don't think so, that variable is not referenced in the code anywhere anymore

vilioren56, 2022-02-21:
Я использую эту переменную, она у меня есть в коде.
Вместо неё надо другую использовать?

mdhiggins, 2022-02-21:
If you update to the latest version of the script on Github its no longer used

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster/commit/d95ddff52ce666e9f3d34d3a7b72f6607b5bd77a

It was replaced with RAWTICK 4 years ago

Just update the script

vilioren56, 2022-02-21:
Instead of USECPERTICK I write RAWTICK?

mdhiggins, 2022-02-21:
Sure why not. All zero instances

closed 2022-02-25 by mdhiggins