mdhiggins/ESP8266-HTTP-IR-Blaster #331: Is it possible to program sequences?

Jeebus678, 2022-03-02:
First of all, thank you for this project. Very impressive. 

My remote doesn't have a singular button that changes input- it requires me to press 'source', up/down arrow, then enter. I've been testing this program out a bit and while it is a beautiful piece of work, I am unsure if this feature is possible.

mdhiggins, 2022-03-02:
You can look up and see if your TV has "discrete" codes which are essentially private codes that do additional functions (like change input, power off without powering on etc)

Other than that aside from programming a sequence of inputs (which the script supports) you might be stuck

mdhiggins, 2022-03-02:
http://www.remotecentral.com/files/use-ir.htm

DragRedSim, 2022-04-02:
I can confirm that it is possible to chain a set of inputs from one command. Here's an example:

`http://example.com:port/json?simple=1&plain=[{'type':'nec','data':'FD0BF4','length':32},{'type':'delay','rdelay':6000},{'type':'nec','data':'FD04FB','length':32},{'type':'nec','data':'FD9867','length':32,'repeat':25},{'type':'nec','data':'FD18E7','length':32,'repeat':1},{'type':'nec','data':'FDA857','length':32}]`

This is an exact command I use on my TV to change to the HDMI 2 input. It turns on the TV (if it isn't on already, since it uses a discrete on/off set), waits six seconds for it to come up, opens the main menu, holds left for 25 presses to scroll to a known good place, presses right once, then presses OK.

closed 2022-04-02 by mdhiggins