mdhiggins/ESP8266-HTTP-IR-Blaster #332: Web service seems to be having issues

DragRedSim, 2022-04-02:
I've been having some issues lately with this package. When speaking to Alexa, there are no commands being sent to the IR blaster device, as shown by the serial output not updating when it should be recieving requests. To help debug this, I attempted to set up a new "other" device with a single command calling to a server I control, to log the request; but the new device isn't showing up in the Alexa app either. Updates to some older devices (changing the name displayed in the Alexa app) also aren't working (the old names still show up). Web requests still work from the local network, as expected, and are blocked from outside (calling the hostname returns HMAC failure); there seems to be a breakdown between the PythonAnywhere service and the Alexa service.

mdhiggins, 2022-04-02:
Try now, seems like there was a backend problem with the lambda skill

DragRedSim, 2022-04-03:
Just tried it again, same results or lack thereof.

mdhiggins, 2022-04-03:
Try logging in and out if the skill. Also you need to rerun device discovery for any changes to take effect regarding device additions or renaming

mdhiggins, 2022-04-03:
Actually looks like they changed around how some of the AWS policy definitions work and there might have been residual issues if you were in the EU or on US-West servers, try now

DragRedSim, 2022-04-04:
Disabled and re-enabled the skill, and things are working now. Thanks for looking into it!

DragRedSim, 2022-04-27:
There seems to be an issue with activating scenes at the moment; commands aren’t being issued when activating them in a routine. The input commands are fine for manual activation, however if a scene is programmed then that will not be called.

mdhiggins, 2022-04-27:
Do scenes work individually? IE not part of a routine?

DragRedSim, 2022-04-28:
When activating a scene by name, eg. “Alexa, turn on Test Scene”, the response is “Test Scene doesn’t support that.” The target of said scene is the same URL as my other commands, just with no parameters in the URL, eg. “http://example.com:port/msg”.

mdhiggins, 2022-04-28:
Looks like Amazon broke version 1.0 SceneController support, migrated to version 3.0 which fixes things

You'll need to go to the pythonanywhere interface and hit "Update Device" to have it migrate to 3.0 and then re-discover devices

DragRedSim, 2022-05-02:
Yep, all working now. Thanks!

closed 2022-05-02 by DragRedSim