mdhiggins/ESP8266-HTTP-IR-Blaster #333: Additional PWM LEDs flicker

Rapti, 2022-04-15:
I'm using this code for a project of mine and really like it. However, I'd like to add an additional simple PWM LED to it, and to avoid needing another microcontroller, I wanted to edit this source code. However, it seems that something is interfering with the ESP8266s PWM output, causing it to flicker. I'm using a D1 Mini and the `mqtt-dev` branch.

To reproduce, put an LED between Pin D3 and GND and add these lines at the top of `void setup()`:

```
  pinMode(0, OUTPUT);
  analogWrite(0, 128);
```
I've also tried using different pins with the same result.
When I upload a minimal sketch with nothing but the above lines in the `setup()`, the flickering stops.
There is also no flickering when I set the duty cycle to 0 or 255, but of course I'd like to use values in-between.

What's causing this? Is there anything I can do?

mdhiggins, 2022-04-15:
I don't have a D1 mini to test against so can't help too much here, its probably another pin interfering, just try switching around the predefined pins and see if it goes away

```
#if defined(ARDUINO_ESP8266_WEMOS_D1R1) || defined(ARDUINO_ESP8266_WEMOS_D1MINI) || defined(ARDUINO_ESP8266_WEMOS_D1MINIPRO) || defined(ARDUINO_ESP8266_WEMOS_D1MINILITE)
const uint16_t  pinr1 = D5;                                          // Receiving pin (GPIO14)
const uint16_t  pins1 = D6;                                          // Transmitting preset 1 (GPIO12)
const uint16_t  configpin = D2;                                      // Reset Pin (GPIO4)
const uint16_t  pins2 = 5;                                           // Transmitting preset 2
const uint16_t  pins3 = 12;                                          // Transmitting preset 3
const uint16_t  pins4 = 13;                                          // Transmitting preset 4
#else
```

That defines the pins that get applied on the d1mini assuming you're using platformIO

Since you don't have a lot of pins available you can try setting all the pins* options to the same pin to free up some options then just cycle them around
That'll let you then shuffle around pinr1 and the configpin

Rapti, 2022-04-16:
Thanks for the reply. I've done that; none of those pins are interfering. I believe the problem is more likely to be caused by a library that you use and not by this project itself.

mdhiggins, 2022-04-16:
Well if you can pinpoint which one let me know and I'll see if I can find a workaround

closed 2022-08-14 by mdhiggins