mdhiggins/ESP8266-HTTP-IR-Blaster #336: Skill offline?

mav2010, 2022-12-05:
Hi,
Is the Alexa skill offline or does it have issues? Since roughly 3 days I am not able to use the skill anymore. It deactivated itself after Alexa saying that she has problems with reaching the light. Whenever I reactivate it and try to switch on/off it doesn't work at all, the Alexa app states "server does not respond".

I have of course tried to delete the light in Alexa and also re-setup the ir blaster instance on the webpage as well. All to no avail.

Any hints if this is central or on my side would be highly appreciated!

Cheers

mdhiggins, 2022-12-05:
Sounds like maybe your local device isn't responding. Invalid passcode means you're not providing the passcode you set up. That same passcode would similarly needed to provide Alexa auth when you set up your URLs. This also gets more complicated if you're using HMAC auth

mav2010, 2022-12-05:
Actually I tested with a NodeMCU where I can easily look at the Serial output - and it worked immediately. My actual device apparently needed a reset (via Pin 10) and a reentering of the Amazon ID again (although it was already shown completely correctly in the corresponding text field). Maybe the flash got corrupted somehow.

What I think I noticed during testing: when the access from Alexa doesn't work because of an apparent mismatch of the Amazon IDs and therewith security code, Alexa appears to disable the Skill immediately. That feels a bit harsh. :-)

But glad that everything is working again. Thanks nonetheless and of course thank you for the great tool and service!