mdhiggins/ESP8266-HTTP-IR-Blaster #337: Issue with NTP Server response

jjx932, 2023-09-18:
Hello I keep getting a no response from the ntp server request. I have swapped out the server from time.google.com to time-a-g.nist.gov however it still returns a no response. Please advise. I dont see my router blocking port 8888 but i may not be looking at it correctly. Everything else works please help.

mdhiggins, 2023-09-18:
It's probably a DNS configuration problem. Can try substituting the IP address for one of those URLs and see if it changes things. Also post the full logs and your configuration

jjx932, 2023-09-22:
attempted with different varying ip addresses of time servers but to the same result.

here are my logs from the serial monitor:
</code>
*wm:STA IP Address: 10.0.1.10
WiFi connected! User chose hostname '' passcode '' and port '80'
WiFi configuration complete
Local IP: 10.0.1.10
DNS IP: 10.0.1.1
URL to send commands: http://ESP-0BF793.local:80Q started
MDNS http service added. Hostname is set to ESP-0BF793.local:80
HTTP Servern port 80
Starting UDP
Local port: 8888
Waiting for sync
Transmit NTP Request
129.6.15.28: 129.6.15.28
�No NTP Response :-(
Turning off the LED to save power.
������Error retrieving external IP
HTTP Code: -1
connection failed
External IP address request took 5001 ms
No errors detected with security configuration
Epoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings
18010
Invalid EPOCH time, security checks may fail if unable to sync with NTP server
Ready to send and receive IR signals

</code>

my configuration is this board v-v with a usb esp flash controller
http://www.chinalctech.com/m/view.php?aid=476

mdhiggins, 2023-09-22:
Does your local IP address configuration make sense? What is your router local IP address and intended subnet From: JJ JayaramanSent: Friday, September 22, 2023 7:33 PMTo: mdhiggins/ESP8266-HTTP-IR-BlasterCc: Michael Higgins; CommentSubject: Re: [mdhiggins/ESP8266-HTTP-IR-Blaster] Issue with NTP Server response (Issue #337) attempted with different varying ip addresses of time servers but to the same result.here are my logs from the serial monitor:*wm:STA IP Address: 10.0.1.10WiFi connected! User chose hostname '' passcode '' and port '80'WiFi configuration completeLocal IP: 10.0.1.10DNS IP: 10.0.1.1URL to send commands: http://ESP-0BF793.local:80Q� startedMDNS http service added. Hostname is set to ESP-0BF793.local:80HTTP Servern port 80Starting UDPLocal port: 8888Waiting for syncTransmit NTP Request129.6.15.28: 129.6.15.28�No NTP Response :-(Turning off the LED to save power.������Error retrieving external IPHTTP Code: -1connection failedExternal IP address request took 5001 msNo errors detected with security configurationEpoch time from timeServer is unexpectedly old, probably failed connection to the time server. Check your network settings18010Invalid EPOCH time, security checks may fail if unable to sync with NTP serverReady to send and receive IR signalsmy configuration is this board v-v with a usb esp flash controllerhttp://www.chinalctech.com/m/view.php?aid=476

jjx932, 2023-09-22:
my router's local address is 192.168.1.1

jjx932, 2023-09-22:
should i change it to a 192.168.x.x type static ip on the device?

mdhiggins, 2023-09-22:
So there's your problem, your DNS IP address is pointing to nothing and your local IP is on a totally different subnet, fix those things or your won't be able to communicate with any network

jjx932, 2023-09-22:
is this the correct section to modify?

char static_ip[16] = "192.168.1.25";
char static_gw[16] = "10.0.1.1";
char static_sn[16] = "255.255.255.0";
char static_dns[16] = "10.0.1.1";

mdhiggins, 2023-09-22:
No, its not modified in code but is set up using WiFiManager during the setup phase, no code edits should be needed

mdhiggins, 2023-09-22:
If you don't already know

https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster#force-wifi-reconfiguration

jjx932, 2023-09-22:
my gpio10 is quite difficult to get to as it is a daughter board soldered onto another, is there a soft way to reset the wifimanager? if not I will try tomorrow when i have my jumpers..

mdhiggins, 2023-09-22:
You could change the `configpin` variable to a different pin and set that other pin to ground during boot

Line 40 and 47

mdhiggins, 2023-09-22:
Also might be able to try

https://www.iottechtrends.com/assets/uploads/2023/02/arduino-ide-how-to-automatically-erase-flash-memory-before-sketch-upload-screenshot.jpg.webp

jjx932, 2023-09-23:
attempted flash wipe didnt work,, will update soon ty for the help :)