mdhiggins/ESP8266-HTTP-IR-Blaster #339: IR from remote received but board won't transmit

warrenjb, 2024-01-23:
Got project loaded to my esp8266 nodemcu and setup with Amazon and router.  Can receive IR signals from TV remote but cannot see LED light up with phone camera on transmit request with Alexa.  Using [these](https://www.amazon.com/gp/product/B07W8W2LNS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) LEDs instead.  Perhaps I need a different resistor setup?  Appreciate any help.

c4coer, 2024-01-28:
hello everybody,
 for same me too.
i am check all it's ok.
only led on D2

mdhiggins, 2024-01-28:
Share the command you're sending

c4coer, 2024-01-28:
hello
here are the commands that I tested:  
Hostname MSG
http://IRCONTROL.local:80/msg?code=C08:SYMPHONY:12
Local IP MSG
http://192.168.0.5:80/msg?code=C08:SYMPHONY:12
Hostname JSON
http://IRCONTROL.local:80/json?plain=[{data:'C08',type:'SYMPHONY',length:12}]

on the controller's web page the orders are indicated sent

thank's

mdhiggins, 2024-01-28:
Do non symphony style codes send? Have you tried modifying the PIN for your sender?

c4coer, 2024-01-28:
for exemple: http://192.168.0.5/json?plain=[{data:%27C08%27,type:%27SYMPHONY%27,length:12,out:4}]
i tested 1 to 4 it's no ok

c4coer, 2024-01-28:
I tried on two D1 mini cards and I use an esp8266 nodemcu V3 because it is more practical, but the problem is the same on both, I do not see any command on the output to drive the transistor

mdhiggins, 2024-01-28:
What's the serial output when you try these commands?

c4coer, 2024-01-28:
I tried D2 and D6

mdhiggins, 2024-01-28:
That does answer the question

And is it any different if you do a non symphony command. Do like a basic SONY or NEC command and see

mdhiggins, 2024-01-28:
Please post the serial output when trying to send these commands

c4coer, 2024-01-28:
actually I tried a sony code and the IR is on.

http://192.168.0.5/msg?code=540C:SONY:15.

so that means that the library, I don't manage my codes?

c4coer, 2024-01-28:
thanks you for your help.

warrenjb, 2024-01-29:
Interesting that the wrong code breaks the device.  Here is my code:
http://externalip:80/msg?code=40040100BCBD:PANASONIC:48&address=0x4004&pass=mypassword
Does this look wrong?

mdhiggins, 2024-01-29:
The air conditioner protocols are often massive and exceed the memory limit of the ESP8266, hence why I was asking about the serial output

The panasonic code looks fine

warrenjb, 2024-01-29:
Any idea why led it not lighting up?  The calls are registering in the cloud.  Also your write up calls for 10 and 1000 resistors but your schematic calls for 100 and 1000.  I assumed the sketch was right?

mdhiggins, 2024-01-29:
Nope post the serial output while you're trying to send a code

And the resistors have quite a bit of flexibility and will vary depending on what LED you've chosen

warrenjb, 2024-01-29:
Serial output:
11:39:04.245 -> Turning off the LED to save power.
11:39:09.570 -> Connection received endpoint '/msg'
11:39:09.570 -> Client IP: X.X.X.X
11:39:09.570 -> Client IP: X.X.X.X
11:39:09.570 -> Passed security check
11:39:09.570 -> MID: 6d6a458e-94ed-4e2a-a5e0-ade3d5bafccf
11:39:09.570 -> Blasting off
11:39:09.570 -> Blocking incoming IR signals
11:39:09.570 -> 40040100BCBD:panasonic:48
11:39:09.616 -> Address: 16388
11:39:09.756 -> Transmission complete
11:39:09.756 -> Reenabling receiving
11:39:10.083 -> Turning off the LED to save power.

mdhiggins, 2024-01-29:
Alright so doesn't look like a software issue. 

Might need to post some pics of your hardware. Also lots of cameras have IR filters so make sure that's not giving you a false negative.

warrenjb, 2024-01-29:
[image]

Pointed remote at phone and saw blast so not a filter on phone issue.

mdhiggins, 2024-02-01:
Looks appropriately wired from what I can tell

Things to try
- use the MQTT-dev branch of the script, this is much newer the main branch is old, it does not have to be used specifically for MQTT; this words best with platformio, default settings should be fine as mqtt is opt-in
[image]
- eliminate the resistors (not a good long term option but for testing purposes you can at least see)
- test your LED with a simple blueprint to ensure the LED is good (can probably find these anywhere, like https://docs.arduino.cc/built-in-examples/basics/Blink/)
- try a simple command without an address (use `A90:SONY:12`)

c4coer, 2024-02-06:
hello
my assembly works well with two LEDs in series. I doubled the reception cache, and I used AI to optimize RAW codes that were too long. thanks you

closed 2024-03-14 by mdhiggins